sylefeb/Silice #41: Don't install from package management if software/tool exists already

dsvf, 2020-06-29:
This tool looks really great!

A slight inconvenience during installation:
When running compile_silice_linux.sh, the installation of e.g. Verilator, yosys etc. is hardcoded and therefore always run.
The repo version of e.g. Verilator in my repo (Linux Mint) is quite old (3.916) but I have a more current version installed. It would be better if the script only installs these tools if the tool exists.
I have no idea how to achieve that and am aware that checking and comparing the version of each tool is a challenge in itself and not the scope of an Algorithm-to-Verilog converter.

sylefeb, 2020-06-29:
Good point, thanks for the feedback. I will add a warning in the documentation at least, and see if there is a simple way to check.

sylefeb, 2020-06-29:
I added checks that verify if the tool is already available on the command line. Hopefully that will avoid this issue.

dsvf, 2020-06-29:
Looks great (and works)! Thanks for the quick fix

sylefeb, 2020-06-29:
Thanks for testing - hope you'll have fun with Silice!

closed 2020-06-29 by sylefeb