sylefeb/Silice #43: Remove unnecessary null pointer checks

elfring, 2020-07-02:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [lua_write_image_in_table](https://github.com/sylefeb/Silice/blob/9817a251d8eb1a0dea8fad74b5c2da2448e7d904/src/LuaPreProcessor.cpp#L201 "lua_write_image_in_table function")
- [ReadTGAFile](https://github.com/sylefeb/Silice/blob/9817a251d8eb1a0dea8fad74b5c2da2448e7d904/src/tga.cpp#L524 "ReadTGAFile function")

sylefeb, 2020-07-03:
Thanks - fixed (even though I do like the explicit checks ; so I hesitated a bit ; but clearly this is well defined behavior, so I'll start cleaning up these cases).

closed 2020-07-03 by sylefeb

elfring, 2020-07-03:
Which commits will contain corresponding adjustments?

sylefeb, 2020-07-03:
https://github.com/sylefeb/Silice/commit/f0b87e6c6e3154acfa94a8b183385122d7060ba0

elfring, 2020-07-03:
Thanks for [your small source code adjustments](https://github.com/sylefeb/Silice/commit/f0b87e6c6e3154acfa94a8b183385122d7060ba0 "Software update").