sylefeb/Silice #52: doom bitstream for ulx3s

emard, 2020-07-28:
HI

I'd like if you could open a separate git repository with
binaries (bitstream, flash image maybe) of this great example:

https://twitter.com/sylefeb/status/1288053013748289538

sylefeb, 2020-07-28:
Hi Emard, thanks for the suggestion - that is a good idea. I am a bit concerned by copyright on the Doom data (as it gets embedded into the bitstream, unsure this is ok or not) ; but I was thinking of using FreeDoom https://freedoom.github.io/ 
I'll look into it, and also will port other demos to the small screen!

emard, 2020-07-28:
HI

For binary repo a free version would be ideal

3D rendering example on 1.3" ST7789 color LCD
is just fantastic

I designed ULX3S, but not imagined
something like this would be easily possible :)

so much great projects appeared recently

sylefeb, 2020-07-28:
This is a great board, a ton of fun to play with, thanks for making this possible! Only getting started on the possibilities ;-)

emard, 2020-07-28:
Thanx! If you have some hardware or support issues,
feel free to ask me. Onboard ESP32 can be used
for remote writing to FPGA, FLASH and SD, faster than USB
and very practical https://github.com/emard/esp32ecp5

I just compiled silice binary, hope that's enough to build something
please add some ULX3S board specific examples -
I like to try compiling blink and 3D checkered pattern
on HDMI or ST7789

sylefeb, 2020-07-28:
Most of the ULX3S work is in the work in progress (wip) branch, with a dedicated folder in projects/build/
https://github.com/sylefeb/Silice/tree/wip/projects/build/ulx3s

Many of the current examples are using a VGA DAC, I am hoping to port to HDMI soon. Also most of the recent work is not yet in the repo. I'll push hopefully tonight (was chasing a few last issues). 

Please let me know if you run into trouble.

emard, 2020-07-28:
Great, I will wait for mainstream push
small hint, option
ecppack --compress will make shorter bitstream which loads much faster

sylefeb, 2020-07-28:
Thanks! I just tested freedoom and ... it worked! Here is a bitstream to test (super low res, just a first attempt) - this is for the ST7789 screen.

[freedoom_e1m1_ulx3s.zip](https://github.com/sylefeb/Silice/files/4991242/freedoom_e1m1_ulx3s.zip)

sylefeb, 2020-07-28:
I also pushed to work in progress ; you could give it a try as it is fairly stable. A good starting point for ULX3S + ST7789 is projects/oled_test.
To build go into projects/build/ulx3s and use
```./ulx3s_bare.sh ../../oled_test/oled_test.ice```
For the Doom-chip, first put doom1.wad in projects/doomchip and then
```./ulx3s_bare.sh ../../doomchip/oled_doomchip.ice``` (takes ~1 hour)

emard, 2020-07-29:
Bitstream works!! tested on ULX3S 85F + ST7789 LCD!
Just a bitstream is required, no FLASH, no SD

Screen is quite dark, has colors, responds fast.
Left button turns right, right button turns left.

Thanx for this incredible example!

sylefeb, 2020-07-29:
Ah great! Thanks for testing it. The image is mirrored left/right, hence the reverted controls :-) (I have to fiddle a bit more with the screen MADCTL) It seems the screen has a Gamma capability, but I could not really get anything out of it yet ...

I believe I have room for significant improvement in both speed and size (LUTs, BRAM is constrained by game data). Working on it!

emard, 2020-07-29:
ST7789 has hardware option to mirror in all 4 directions

Here are my initialization files st7780_linit_*.mem
They differ in MADCTL CASET RASET parameters

https://github.com/emard/ulx3s-misc/tree/master/examples/spi_display/hdl/spi_display_verilog

emard, 2020-07-29:
this init sequence contains control bytes that are not sent to SPI
but they describe for number of parameters that follow is is there
delay after the command

I tried to comment them in one of the files like this
https://github.com/emard/ulx3s-misc/blob/master/examples/spi_display/hdl/spi_display_verilog/st7789_linit.mem#L24