sylefeb/Silice #55: Framework for FOMU

rob-ng15, 2020-08-28:
Hello there,

I'm trying to put together a framework for the FOMU, a simple USB FPGA, as Silice looks like a nice way to play with FPGA programming. Any advice?

I'll post what I've tried so far when at the computer, but as of yet I cannot get the blink (first example) to work.

Many thanks,

Rob.

sylefeb, 2020-08-28:
Hello,

Sounds great - maybe look at the icestick or ulx3s frameworks as examples. Adding a board currently requires two things:

1) The most important is the verilog framework, they are all located in the 'framework' directory, e.g.:
https://github.com/sylefeb/Silice/blob/master/frameworks/icestick.v
Note there are a few defines at the top, both verilog and Silice, which are convenient to do multi-board projects.

2) I also create build scripts in 'projects/build', for instance for the icestick:
https://github.com/sylefeb/Silice/tree/master/projects/build/icestick
This depends a bit on the board and toolchain. For instance for the icestick there is a pcf file for the pin specification in nextpnr-ice40. For other board/tools other extra-files may be required.

So far I've been creating different frameworks depending on the board features (e.g. if an OLED screen is plugged, or SDRAM is available), but I'd like to come up with a better system. I'd also like to look into https://github.com/olofk/edalize which would provide a more unified approach. Advice welcome!

If you have any more trouble let me know, happy to help.

rob-ng15, 2020-08-28:
Using this as a framework (fomu.v):

``define FOMU 1
`default_nettype none
`define BLUEPWM  RGB0PWM
`define GREENPWM RGB1PWM
`define REDPWM   RGB2PWM
$$FOMU=1
$$HARDWARE=1

module top(
  // 48MHz Clock Input
  input   clki,
  // LED outputs
  output rgb0,          // blue
  output rgb1,          // green
  output rgb2,          // red
  // USB Pins
  output usb_dp,
  output usb_dn,
  output usb_dp_pu,
  // SPI
  output spi_mosi,
  input  spi_miso,
  output spi_clk,
  output spi_cs,
  // USER pads
  input user_1,
  input user_2,
  input user_3,
  input user_4
);

    // Assign USB pins to "0" so as to disconnect Fomu from
    // the host system.  Otherwise it would try to talk to
    // us over USB, which wouldn't work since we have no stack.
    assign usb_dp = 1'b0;
    assign usb_dn = 1'b0;
    assign usb_dp_pu = 1'b0;

    // Connect to system clock (with buffering)
    wire clk;
    SB_GB clk_gb (
        .USER_SIGNAL_TO_GLOBAL_BUFFER(clki),
        .GLOBAL_BUFFER_OUTPUT(clk)
    );

    wire [2:0] __main_led;

    SB_RGBA_DRV #(
        .CURRENT_MODE("0b1"),       // half current
        .RGB0_CURRENT("0b000011"),  // 4 mA
        .RGB1_CURRENT("0b000011"),  // 4 mA
        .RGB2_CURRENT("0b000011")   // 4 mA
    ) RGBA_DRIVER (
        .CURREN(1'b1),
        .RGBLEDEN(1'b1),
        .`BLUEPWM(__main_led[0]),     // Blue
        .`REDPWM(__main_led[1]),      // Red
        .`GREENPWM(__main_led[2]),    // Green
        .RGB0(rgb0),
        .RGB1(rgb1),
        .RGB2(rgb2)
    );

reg [31:0] RST_d;
reg [31:0] RST_q;

reg ready = 0;

always @* begin
  RST_d = RST_q >> 1;
end

always @(posedge clk) begin
  if (ready) begin
    RST_q <= RST_d;
  end else begin
    ready <= 1;
    RST_q <= 32'b111111111111111111111111111111;
  end
end

wire reset_main;
assign reset_main = RST_q[0];
wire run_main;
assign run_main = 1'b1;

M_main __main(
  .clock        (clk),
  .reset        (RST_q[0]),
  .out_led      (__main_led),
  .in_run       (run_main)
);

endmodule
`
And this as first_example.ice

`algorithm main(output uint3 led) {   
  uint28 counter = 0;      // a 28 bits unsigned integer
  led := counter[25,3];    // LEDs updated every clock with the 8 most significant bits  
   while (1) {              // forever
    counter = counter + 1; // increment counter
  }  
}
`
I can get a set of colours flashing on the FOMU. I've not idea if the RGB led is actually assigned in the correct order, but just pleased that something works!

I used the code in the fomu-workshop from blink.v (https://github.com/im-tomu/fomu-workshop/blob/master/verilog/blink/blink.v) as a guide.

Rob.

sylefeb, 2020-08-28:
That's great, thanks!
To check the order, since this is a counter, the blue should blink twice as fast as the red which then blinks twice as fast as the green (the most significant bits). If the blinking is too fast, make the counter wider and adjust counter[25,3] accordingly (which means 3 bits from bit 25 as you have likely guessed already ;) )

rob-ng15, 2020-08-29:
Hi there,

Have tweaked this slightly, so I believe the lights are now in the correct order. I have also generated a framework that incorporates a USB_ACM, which is useful for communication to/from the FOMU. I'll ask permission from the original author if that can be included, and will forward those as well.

Rob.

rob-ng15, 2020-08-30:
Hi there,

I've got things working, both an LED and a UART example. I've uploaded them to https://github.com/rob-ng15/Silice-Playground

Rob.

sylefeb, 2020-08-30:
Thanks for this! Should I wait before merging the PR? (I believe it does not incorporate the changes mentioned above?)

rob-ng15, 2020-08-30:
Hi there,

It might be better to just use these?

Rob.

sylefeb, 2020-08-31:
Hi Rob, ok will integrate later today!

sylefeb, 2020-08-31:
PR is closed?

rob-ng15, 2020-08-31:
Apologies, I think I pressed the wrong button on the wrong commit. The
version from the playground repository is the most up-to-date.

Thanks for planned merger.

Rob.

sylefeb, 2020-08-31:
Great, thanks!

sylefeb, 2020-08-31:
I started integration in the draft branch:
https://github.com/sylefeb/Silice/tree/draft/projects/build/fomu

Seems ok? 

I have to polish a few things there and then will proceed to merge draft=>wip=>master hopefully in the next couple days. 
Thanks again for making this! I'll try to grab a FOMU soon, looks like a nice tiny board.

sylefeb, 2020-09-01:
Done => closing for now

closed 2020-09-01 by sylefeb

rob-ng15, 2020-09-01:
Thanks for that! Appreciated.

Yes the FOMU is an interesting idea, I was given one for my contribution to
litex, but I couldn't work out what to do with it as I had no simple way to
get information in or out, the USB stack I used from Stef and TinyFpga
certainly helps.

There seems to be a lot of FOMU users interested in Silice, so hopefully
you'll get a lot of users from this, the FOMU workshop certainly want to
direct people to this.

Are you planning on implementing a switch or select-case style notation?
Would make the demo code a little clearer.

I have a de10nano too, hence my interest in litex, and Silice, and now that
case sensitivity is fixed I'll be able to try out the doomchip.

Thanks for your great work, Silice has certainly given people like myself
an easy way in to FPGA programming.

Rob.

sylefeb, 2020-09-01:
Hi Rob - please let me know if you have any other trouble with the doomchip. It takes a long time to synthesize, so plan for some tea/coffee ;)

> Are you planning on implementing a switch or select-case style notation?

Actually there is already a switch-case statement (documentation is lagging behind, have to update it!). Here is an example:
https://github.com/sylefeb/Silice/blob/27230577638a929beaffa4db940aedd5482fa0aa/projects/ice-v/ice-v.ice#L478
(there are several others in this file)

Thanks!

rob-ng15, 2020-09-01:
Switch Case is perfect, I'll update the usb_acm example to use it.

Rob.

rob-ng15, 2020-09-01:
Hi there,

That was easy, apart from I'm English, we spell uninitialized as
uninitialised! Could be worth aliasing for us!

I've updated my playground example, and as a bonus it saves a few LUTS.

Rob.

rob-ng15, 2020-11-28:
> In the linux version I wish to install in the default directory i.e.
> /usr/local I have done it manually but why does not the scripts work?

I don't know, you'll have to ask @sylefeb/Silice
<reply@reply.github.com>  the
developer of Silice. He has an issue tracker on github.