sylefeb/Silice #63: icestick.pcf always marked as modified

trabucayre, 2020-09-15:
Each time I clone Silice repository `projects/build/icestick/icestick.pcf` is marked as modified.
`git diff` output is a bit strange because it's seems all lines are reformated (unix, dos, mac?).
 
It's maybe a problem with all my computers (or with my distribution). I have tried with debian, centos and gentoo. In all case the result is the same.

sylefeb, 2020-09-15:
That is odd. I configured the eol explicitly to be lf style in the gitattributes:
https://github.com/sylefeb/Silice/blob/master/.gitattributes
That should cover all cases?

I mainly use cygwin under Windows. I sometimes see issues with file permissions (between MinGW/cygwin) but not with autocrlf, including on my Linux systems (just retried on an Ubuntu).

trabucayre, 2020-09-16:
A quick test with some file show all of them are lf.
It's not true for `icestick.pcf`:
```
$ file icestick.pcf 
icestick.pcf: ASCII text, with CRLF line terminators
```
This is confirmed by reading `hexedit` content in hex format where 0x0D 0x0A

If you want I can convert this file in unix format ?

sylefeb, 2020-09-16:
Nice catch - thanks for the fix!

closed 2020-09-16 by sylefeb