sylefeb/Silice #81: ENHANCEMENT: Request

rob-ng15, 2020-10-07:
```
    brom uint16 rom[] = {
        $include('j1eforthROM.inc')
    };
    
    dualport_bram uint16 ram[32768] = uninitialized;
```
Combined to : 

```
dualport_bram uint16 ram[32768] = {
        $include('j1eforthROM.inc')
    };
```
to remove the need to copy the rom to the working ram area.

Will free up bram/brom space in the j1eforth design, and bring that area of Silice into equivalence with verilog, which I believe allows something similar.

sylefeb, 2020-10-07:
Good point, thanks. Will likely make it explicit with a syntax to say 'pad the rest with value X'

sylefeb, 2020-10-07:
Maybe like so:
```
dualport_bram uint16 ram[32768] = {
        $include('j1eforthROM.inc')
        , pad(0)
    };
```

rob-ng15, 2020-10-07:
Excellent idea.

rob-ng15, 2020-10-07:
That would be exactly what I was thinking.

rob-ng15, 2020-10-08:
This is working perfectly, for pad(uninitialized) and for pad(value) blocks.

closed 2020-10-08 by rob-ng15