sylefeb/Silice #98: Build for ULX3S failes

goran-mahovlic, 2020-11-24:
Hi!
First thank you on your amazing work!

I want to try latest wip branch ulx3s build, but it constantly files with message
Do you maybe know how to fix this one. 
I think I have recent opensource toolchain.

Using template $paramod$30bfbdd93cc1df27849a2af1e8b962fe07f8e9db\$__ECP5_DP16KD for cells of type $__ECP5_DP16KD.
/media/internal/FPGA/Silice/frameworks/boards/ulx3s/ulx3s.sh: line 35: 21066 Killed                  yosys -p 'scratchpad -copy abc9.script.flow3 abc9.script; synth_ecp5 -abc9 -json build.json' build.v
ERROR: Failed to open JSON file 'build.json'.
0 warnings, 1 error

Or if you could send me bitstream for freedoom on mail goran.mahovlic@gmail.com
In two day Radiona have small gaming dedicated exhibition in technical museum and we want to show your work there.
Best bitstream would be the one where you cannot shoot - "Pacifistic FreeDOOM" :)

sylefeb, 2020-11-24:
Hi Goran, thanks! I can prepare a bitstream (without a gun :) ) of the latest (not yet in Silice repo) ; I can also send you an sdcard image to go along with it. Would you like to use data from the original doom or the freedoom project? Any preferred level?

Let's try to fix the issue you are observing. Could you please send me the version strings of yosys and nextpnr from your system? (I have not updated them here in ~ 1 month). I'll also rebuild from scratch and check for issues here asap.

goran-mahovlic, 2020-11-24:
Tnx!

Any no shooting version will be fine, and any level - the story of your project is what makes it special!
Yes prepared SD card image would be great!

Yosys 0.9+2406 (git sha1 bbf34358, clang 7.0.1-8 -fPIC -Os)
nextpnr-ecp5 -- Next Generation Place and Route (Version 3eff12a)

sylefeb, 2020-11-24:
Here I am using 
Yosys 0.9+3537 (git sha1 325d88b0, gcc 10.2.0 -fPIC -Os)
nextpnr-ecp5 -- Next Generation Place and Route (Version d5dde5df)

Just launched a rebuilt from wip (using 'make ulx3s' in projects/doomchip after rebuilding silice), looking ok so far (but let's see how it ends in ~ 20 minutes).

goran-mahovlic, 2020-11-24:
Great! 

I will probably need to update oss

I am mostly using this repo for releases

https://github.com/alpin3/ulx3s/releases/tag/v2020.10.12

but I could try to get newer from here 

https://github.com/open-tool-forge/fpga-toolchain/releases

sylefeb, 2020-11-24:
It worked! (sending by email as this older version carries the game data - even though that is from the widely available doom1.wad).

sylefeb, 2020-12-02:
I can confirm the default Ubuntu packages are outdated. After re-building yosys, trellis and nextpnr (following the instructions on their respective githubs), everything works properly. I am going to close the issue but please let me known if something is still wrong after using the latest versions.

closed 2020-12-02 by sylefeb