sylefeb/Silice #101: cannot build blinky for ulx3s, errors

arblake, 2020-11-29:
The build process fails here:
1. Executing Verilog-2005 frontend: build.v
Parsing Verilog input from `build.v' to AST representation.
Storing AST representation for module `$abstract\top'.
Storing AST representation for module `$abstract\M_main'.
Successfully finished Verilog frontend.

Syntax error in command `synth_ecp5 -nomux -abc9 -top top':

    synth_ecp5 [options]

This command runs synthesis for ECP5 FPGAs.

Assistance required

sylefeb, 2020-11-29:
Hi,

I assume you are running 'make ulx3s' from command line in the projects/blinky folder, correct?
Could you give me the version strings of your system yosys and nextpnr? Here I am using:

Yosys 0.9+3537 (git sha1 325d88b0, gcc 10.2.0 -fPIC -Os
nextpnr-ecp5 -- Next Generation Place and Route (Version d5dde5df)

Also which variant of the ULX3S do you have? (the current framework is tested on 85F, I have not done the other versions as I do not own the hardware, but it should be easy once we can test).

arblake, 2020-11-30:
Yosys 0.9 (git sha1 1979e0b)
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
nextpnr-ice40 -- Next Generation Place and Route (Version b5054f0d)
Yes,    make ulx3s
ulx3s with ECP5 85F

Adrian

sylefeb, 2020-11-30:
Ok thanks, I am installing an Ubuntu 20.04 to test on the same config, looking into it!

sylefeb, 2020-12-02:
Alright, so I can confirm the default Ubuntu packages are outdated. After re-building yosys, trellis and nextpnr (following the instructions on their respective githubs), everything works properly. I leave the issue open, could you please try and confirm?
(I recommend removing the packages, e.g. "sudo apt remove yosys" before doing the compilation and install from source)

sylefeb, 2020-12-18:
I will close the issue as this has been tested on latest version - please let me know in case the problem persists.

closed 2020-12-18 by sylefeb