sylefeb/Silice #102: Issue with ecp5 BRAM inference on latest yosys

sylefeb, 2020-12-03:
With the latest Yosys the brams are no longer inferred properly. I am investigating the issue.

Version Yosys 0.9+3537 (git sha1 c75d8c74, gcc 10.1.0 -fPIC -Os) is known to work.

sylefeb, 2020-12-03:
The issue occurs in both `2.27` and `2.29` MEMORY_BRAM pass.

What I found so far:
- On a bram, wenable must be set if unused, e.g. wenable := 0 ; Silice then creates a register always assigned 0 (btw it should not since constant, corner case not detected by optimizer...) which leads to an inferred bram. 
So, assigning a constant (wire assigned 0) to the write port puts it in ~async~ while it previously wouldn't.

- A dualport_bram no longer works, as it is no longer detected as simple dual port even when used as such (wenable0 := 0, wenable1 := 1) ; however editing the template to force them as simple dualport seems to work
This last point relates to #91. I might have to introduce an explicit simple_dualport_bram.

rob-ng15, 2020-12-04:
Version Yosys 0.9+3617 (git sha1 fd306b05, gcc 10.2.0 -march=x86-64
-mtune=generic -O2 -fno-plt -fPIC -Os) also works.

sylefeb, 2020-12-04:
A workaround prototype is pushed on draft (https://github.com/sylefeb/Silice/commit/e5c7e7b053fdc02b37d641687e51b658004915b5). 
As suggested in https://github.com/YosysHQ/yosys/issues/2473, forcing flip-flops on bram inputs leads to correct inference. 

Also resolving #91 by introducing an explicit simple_dualport_bram. This means, however, that code using dualport_bram to implicitly create a simple_dualport_bram (through a constant choice of wenable0/wenable1) now has to switch to simple_dualport_bram  for correct synthesis. Silice will return an error message if using a dualport_bram when not supported by the target hardware. Example projects are being updated and tested. The simple_dualport_bram reads on port 0 and writes (if wenable1 == 1) on port 1.

rob-ng15, 2020-12-05:
Certainly works for my Risc-ICE-V CPU on yosys-git r10262. The result for yosys-git r10391 does not properly work. Not sure if this is due to an inherent flaw in my design though.

rob-ng15, 2020-12-16:
I think that after the testing this has received, certainly on the ULX3S, with various versions of yosys, including `Yosys 0.9+3758 (git sha1 5a881497e, gcc 10.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fPIC -Os)` this can be closed.

sylefeb, 2020-12-16:
Thanks for your help with this!

closed 2020-12-16 by sylefeb