sylefeb/Silice #104: build error, can't read build.config

arblake, 2020-12-30:
Trying to build j1eforth for the ulx3s.
Good progress until :
2.51. Executing JSON backend.

End of script. Logfile hash: 9bd1872882, CPU: user 2.93s system 0.11s, MEM: 265.32 MB peak
Yosys 0.9+3715 (git sha1 1c4a18f6, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)
Time spent: 24% 1x abc9_exe (0 sec), 19% 20x read_verilog (0 sec), ...
Warning: net 'clk_25mhz' does not exist in design, ignoring clock constraint
ERROR: IO 'clock' is unconstrained in LPF (override this error with --lpf-allow-unconstrained)
1 warning, 1 error
sed: can't read build.config: No such file or directory
Failed to open input file
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built Dec  3 2020 08:47:16)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 85K v3.0.8
open(build.bit) failed

The problem may be easily solved if I knew where to look.
Adrian

rob-ng15, 2020-12-31:
Hi Adrian,

I've just tried building with `Yosys 0.9+3803 (git sha1 449154803, gcc
10.2.0 -march=x86-64 -mtune=generic -O2 -fno-plt -fPIC -Os)
` and it completes. Not sure why you are getting the error you are getting
though.

I have some backporting/improvements to copy over from PAWS/Risc-V once
I've got that in a more stable state. Particularly to the GPU.

Rob.

arblake, 2020-12-31:
Rob,
I upgraded yosys : Yosys 0.9+3819 (git sha1 48d0aeb0, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)

Same error. Where is yosys called and there I may be able to add "--lpf-allow-unconstrained"  which appears to be the problem.
Is there a configuration file where the clock is set to 25MHz and should be 50MHz?
If I can get this to build I will explain my reasons.
Adrian

rob-ng15, 2020-12-31:
I've no idea on the answer to your question. I am always battling with
yosys!

arblake, 2020-12-31:
Rob,
I think the problem is in frameworks. But I cannot recall how to find and make frameworks

arblake, 2020-12-31:
FIXED ))  All related to  path to frameworks

closed 2020-12-31 by arblake

rob-ng15, 2021-01-02:
BTW Adrian,

I have backported the changes to the BITMAP and GPU layer to j1eforth, and
I've also improved the CPU with CALLBRANCH and ALU units.

Rob.