sylefeb/Silice #106: Extent of AGPL licensing

sftwninja, 2021-01-01:
I wanted to play around with Silice and potentially make my own things. When licensing Silice under the AGPL, was it intended that the built artifacts also inherit that AGPL, or does that only apply to the Silice toolsuite itself?

From a 10,000 meter view, I don't know what is ultimately "linked" into an end target, so I wanted to make sure anything I made wasn't potentially infringing.

sylefeb, 2021-01-02:
Hi - my intent was to only cover Silice itself and the designs that come with it (so the '.ice' designs in this repo are all under the same AGPL license by default).

If you write your own code and use Silice to generate Verilog from it, only a few things are appended to that generated Verilog code (basically the framework glue, and the templates which are used to generate e.g. the BRAM modules). I wouldn't want this to propagate the AGPL to the entire produced design. Should the license of these specific files change for something more permissive?

But in any case, to state it clearly the intent is:
- Silice source code, accompanying examples and everything in the repo is AGPL
- The AGPL does not extend/propagate to what you produce with your own code

Please do great things with Silice and let us know about it!

sylefeb, 2021-06-22:
Discussion on HN with feedback regarding AGPL
https://news.ycombinator.com/item?id=27575174