sylefeb/Silice #107: Defines in board.json do not seem to be applied

TaraHoleInIt, 2021-01-09:
Hi.

I'm working on setting up a framework for the TinyFPGA BX, and while setting up GPIOs I noticed that despite having GPIO=1 in board.json the verilog file will not see it as being set.

Putting this at the start works, but then all examples require main to implement the gpio arguments.
This seems to apply to the ulx3s as well.

```
`define GPIO 1
```

Output of building blinky with ``` `defineGPIO 1``` in tinyfpga_bx.v:
```
2.3.3. Analyzing design hierarchy..
Top module:  \top
Used module:     \M_main
ERROR: Module `M_main' referenced in module `top' in cell `__main' does not have a port named 'in_gn'.
make[2]: *** [build.mk:9: build.blif] Error 1
make[2]: Leaving directory '/home/tara/Silice/projects/blinky/BUILD_tinyfpga_bx'
make[1]: *** [Makefile:21: build.json] Error 2
make[1]: Leaving directory '/home/tara/Silice/projects/blinky/BUILD_tinyfpga_bx'
Traceback (most recent call last):
  File "/home/tara/.local/lib/python3.8/site-packages/edalize/edatool.py", line 382, in _run_tool
    cp = run([cmd] + args,
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make']' returned non-zero exit status 2.
```

tinyfpga_bx.v: https://gist.github.com/TaraHoleInIt/7ddd201f32b588af83293c01b2801fa1
board.json: https://gist.github.com/TaraHoleInIt/6b2012e8f4867f8dbff687f28d2c2ca2

sylefeb, 2021-01-09:
Hi - thanks for working on this! 

The defines will only be added if the Makefile asks for the set of pins, see for instance
https://github.com/sylefeb/Silice/blob/367ae5ca4f4ff7b155ec84c518fa647b8242eb35/projects/oled_sdcard_test/Makefile#L4
The parameter `-p basic,oled,sdcard,buttons` tells the Silice build system that these pins are requested by the project. This will select the corresponding defines for both the framework and Silice preprocessor.

On Silice side, when writing main, you can use the pre-processor defines to select the pins, depending on this Makefile parameter:
https://github.com/sylefeb/Silice/blob/367ae5ca4f4ff7b155ec84c518fa647b8242eb35/projects/common/video_sdram_main.ice#L234-L240
(in this case this also changes the design depending on whether sdcard, hdmi, vga or audio are present)

So basically the selection always starts from the Makefile parameters, this triggers the defines for both the framework and the Silice preprocessor. I'd like to make writing the main easier in the future (either when starting a new project, or with a preprocessor var, e.g. $pins$, unsure what would be best).

Btw, I started to document the process of assing a new board, this is just a draft but I though it might help:
https://github.com/sylefeb/Silice/tree/draft/frameworks/boards

TaraHoleInIt, 2021-01-10:
Ohhhh that makes perfect sense.

I'll have to see if I can make a proper patch to submit once I have more things tested out.
I wasn't sure how to handle GPIO so I decided to have the ones on the left be outputs and the ones on the right be inputs.

I was also wondering if the verilog file should have definitions for frequency, period, and other such constants or if they are best left in individual files that need them?

sylefeb, 2021-01-11:
GPIO: ah, yes there is a similar situation on the ULX3S ; I'll have to come up with something so we can choose at design time (I tried with inout but my first attempt was not convincing).

*definitions for frequency, period* => That is a good point ; the base frequency at least should be there. Maybe a CLOCK_FREQ_MHZ define ?

TaraHoleInIt, 2021-01-11:
Having the clock frequency should be enough, the rest can be calculated easy enough if needed.

As an aside:
What is the best way to ask about syntax and code problems without being overly annoying?

sylefeb, 2021-01-11:
Please feel free to ask! This will also help building up resources for everyone. How about we start with an always open issue here? (with a title inviting for such questions?). When/if that becomes unmanageable maybe a Google group? But I'd like to have everything here if possible.

closed 2021-01-14 by sylefeb