sylefeb/Silice #114: no build.json

arblake, 2021-02-18:
IN attempting to build I receive this error:

2.44.15.3. Executing XAIGER backend.
ERROR: Assert `!aig_map.count(bit)' failed in backends/aiger/xaiger.cc:435.
ERROR: Failed to open JSON file 'build.json'.
0 warnings, 1 error

Where is it ? and where should it be?

sylefeb, 2021-02-20:
Hi - thanks for the report - could you give me a bit more context? which board are you compiling for? if possible please send me a reproduction sequence so I can trigger the issue here. Thank you!

sylefeb, 2021-02-20:
It could be a side effect of compilation failing earlier, leading to build.json not being output, maybe check the yosys logs? should be in BUILD_*/yosys.log. Any error message there?

arblake, 2021-02-20:
Hi,

The board is ULX3S

"

2.44.15.1. Executing ABC9_OPS pass (helper functions for ABC9).

2.44.15.2. Executing ABC9_OPS pass (helper functions for ABC9).

2.44.15.3. Executing XAIGER backend.
ERROR: Assert `!aig_map.count(bit)' failed in backends/aiger/xaiger.cc:435.
ERROR: Failed to open JSON file 'build.json'.
0 warnings, 1 error
sed: can't read build.config: No such file or directory
 

"

I hope this helps
The project is being built using silice-make.py and I do not know where the yosys logs are located.

Adrian

arblake, 2021-02-24:
Looks like a yosys error but no fix

rob-ng15, 2021-02-24:
Hi Adrian,

I've tested:

yosys-git-0.9+3905.r10550 and yosys-git-0.9+3911.r10560

they both give the error you describe. I've held at

yosys-git-0.9+3833.r10478

as this works.

Rob.

sylefeb, 2021-02-24:
Hi Adrian, Hi Rob,

How can I reproduce the problem? (Mostly away from keyboard this week but will look into this with high priority). I assume this is when compiling Rob's PAWS?

Thanks!
Sylvain

arblake, 2021-02-24:
Sylvain
Not PAWS but j1eforth on ULX3S
Adrian

rob-ng15, 2021-02-24:
Hi,

The problem also occurs with PAWS using an update yosys beyond version 3833.

Rob.

arblake, 2021-03-02:
Hi,
Any additional thoughts on this problem ?
Adrian

sylefeb, 2021-03-02:
Hi Adrian, not yet, hoping to start looking into it soon. Meanwhile please use the older version of yosys if possible, it may take a bit of time to understand + fix depending on the issue.

arblake, 2021-03-02:
Hi,
How do I get an older version?
I only know the basics of git.
Adrian

-- 
Adrian Blake, VK2ALF
101 Mulach St
Cooma, NSW, 2630
Australia
Mobile +61 407232978

Pärna tn 3
Otepää
Valgamaa
Estonia
Mobile +372 51971441

sylefeb, 2021-03-02:
If you are compiling yosys from source (highly recommended) then it is a matter of identifying a version checkout. For instance, @rob-ng15 pointed out that version yosys-git-0.9+3833.r10478 is fine. This is determined by YOSYS_VER in the Makefile, so looking at history (https://github.com/YosysHQ/yosys/commit/b0004911ca15ec41db56c76c83461061fedc742e#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52) this is commit b0004911ca15ec41db56c76c83461061fedc742e

You can the do `git checkout b0004911ca15ec41db56c76c83461061fedc742e` to obtain the right version, and recompile (see README instructions https://github.com/YosysHQ/yosys )

sylefeb, 2021-03-20:
Started work on this. So it might be that there is a combinational loop in the design, which then triggers a known issue in yosys (https://github.com/YosysHQ/yosys/issues/2530). We should remove the combinational loop anyway (these are usually bad and don't do what was intended). Silice detects many of them but not all  yet, unfortunately.

To try to detect the loops I hacked a verilator version of j1eforth. Verilator usually catches the combinational loops indicating a Warning-UNOPTFLAT. Here is what I get so far:
```
%Warning-UNOPTFLAT: build.v:5467:13: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main.CALLBRANCH._d_newStackTop'
 5467 | reg  [15:0] _d_newStackTop,_q_newStackTop;
      |             ^~~~~~~~~~~~~~
                    build.v:5467:13:      Example path: top.__main.CALLBRANCH._d_newStackTop
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34271:13:      Example path: top.__main._d_stackTop
                    build.v:5497:1:      Example path: ALWAYS
                    build.v:5467:13:      Example path: top.__main.CALLBRANCH._d_newStackTop
%Warning-UNOPTFLAT: build.v:5468:13: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main.CALLBRANCH._d_newPC'
 5468 | reg  [12:0] _d_newPC,_q_newPC;
      |             ^~~~~~~~
                    build.v:5468:13:      Example path: top.__main.CALLBRANCH._d_newPC
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34271:13:      Example path: top.__main._d_stackTop
                    build.v:5497:1:      Example path: ALWAYS
                    build.v:5468:13:      Example path: top.__main.CALLBRANCH._d_newPC
%Warning-UNOPTFLAT: build.v:5469:12: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main.CALLBRANCH._d_newDSP'
 5469 | reg  [7:0] _d_newDSP,_q_newDSP;
      |            ^~~~~~~~~
                    build.v:5469:12:      Example path: top.__main.CALLBRANCH._d_newDSP
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34273:12:      Example path: top.__main._d_dsp
                    build.v:5497:1:      Example path: ALWAYS
                    build.v:5469:12:      Example path: top.__main.CALLBRANCH._d_newDSP
%Warning-UNOPTFLAT: build.v:5470:12: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main.CALLBRANCH._d_newRSP'
 5470 | reg  [7:0] _d_newRSP,_q_newRSP;
      |            ^~~~~~~~~
                    build.v:5470:12:      Example path: top.__main.CALLBRANCH._d_newRSP
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34287:12:      Example path: top.__main._d_rsp
                    build.v:5497:1:      Example path: ALWAYS
                    build.v:5470:12:      Example path: top.__main.CALLBRANCH._d_newRSP
%Warning-UNOPTFLAT: build.v:5151:13: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main.ALU.ALUplus._d_newStackTop'
 5151 | reg  [15:0] _d_newStackTop,_q_newStackTop;
      |             ^~~~~~~~~~~~~~
                    build.v:5151:13:      Example path: top.__main.ALU.ALUplus._d_newStackTop
                    build.v:5392:1:      Example path: ALWAYS
                    build.v:5342:13:      Example path: top.__main.ALU._d_newStackTop
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34271:13:      Example path: top.__main._d_stackTop
                    build.v:5172:1:      Example path: ALWAYS
                    build.v:5151:13:      Example path: top.__main.ALU.ALUplus._d_newStackTop
%Warning-UNOPTFLAT: build.v:4969:13: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main.ALU.ALU._d_newStackTop'
 4969 | reg  [15:0] _d_newStackTop,_q_newStackTop;
      |             ^~~~~~~~~~~~~~
                    build.v:4969:13:      Example path: top.__main.ALU.ALU._d_newStackTop
                    build.v:5392:1:      Example path: ALWAYS
                    build.v:5342:13:      Example path: top.__main.ALU._d_newStackTop
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34273:12:      Example path: top.__main._d_dsp
                    build.v:4990:1:      Example path: ALWAYS
                    build.v:4969:13:      Example path: top.__main.ALU.ALU._d_newStackTop
%Warning-UNOPTFLAT: build.v:34259:13: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main._d_pc'
34259 | reg  [12:0] _d_pc;
      |             ^~~~~
                    build.v:34259:13:      Example path: top.__main._d_pc
                    build.v:34467:21:      Example path: ASSIGNW
                    build.v:34248:14:      Example path: top.__main._w_pcPlusOne
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34259:13:      Example path: top.__main._d_pc
%Warning-UNOPTFLAT: build.v:34257:13: Signal unoptimizable: Feedback to clock or circular logic: 'top.__main._d_instruction'
34257 | reg  [15:0] _d_instruction;
      |             ^~~~~~~~~~~~~~
                    build.v:34257:13:      Example path: top.__main._d_instruction
                    build.v:34458:23:      Example path: ASSIGNW
                    build.v:34245:13:      Example path: top.__main._w_rstackWrite
                    build.v:34474:1:      Example path: ALWAYS
                    build.v:34257:13:      Example path: top.__main._d_instruction
```

sylefeb, 2021-03-20:
I'll try to dive a bit deeper into j1eforth but I'll need @rob-ng15 help to work on a fix.

sylefeb, 2021-03-20:
The source of problems are some of the `output!` and expression trackers `uint1 pcPlusOne := pc + 1` in `j1eforth.ice`. I have long postponed a write up on that, will do asap. Basically both these constructs are prone to lead to combinational loops.

`output!` is cabling the value being modified during the cycle to the output (the 'd' side of the flipflop). If this is used to drive any of the inputs bound with `<:` a loop is created. Silice detects some, but not all, and these are to be considered as errors. This can be fixed by either binding the dependent input with `<::` or using `output`. But of course, this introduces a one-cycle latency either on the input or output. Again, a cycle exists only if there is a dependency between the output and one of the input, otherwise that would be ok.

When creating an expression tracker (wire) with `:=`, similarly the value being modified during the cycle is tracked. That means cyclic expressions can quickly be created if the tracker is used to modify anything onto which the right hand side depends. One such case exists for `uint13  pcPlusOne ::= pc + 1;`, because `pc` may be assigned `newPC` and `newPC` itself depends on `pcPlusOne`. This can be fixed by using `::=` which means the values at the cycle start are tracked from the right side (here, `pc` as it was at cycle start) or by changing the logic if possible to break the cycle.

sylefeb, 2021-04-04:
Hello, I have improved combinational loop detection significantly (#120) but I am unsure this resolved the issue at end. I believe @rob-ng15 now has a version that compiles on latest yosys? Shall we keep the issue open?

sylefeb, 2021-04-04:
(regarding my report above using Verilator, I now suspect some of these are false positive due to using $display, but not fully confident).

rob-ng15, 2021-04-04:
Both j1eforth and PAWS build with yosys 0.9+4052.r10702.0ccc7229c-1, so it looks like there was a bug in yosys, which was still present in 4008 the last version I tried.

So whilst the issue at hand seems to be fixed, this could be closed, unless you believe there are further improvements to be made to the combinatorial loop detection in Silice?

rob-ng15, 2021-04-04:
Now wip has been updated I can update my Silice and try building tomorrow. Will I need to start moving unit8 X := y type expressions to <: ?

arblake, 2021-04-05:
Some progress. Error is now:
2.51. Executing JSON backend.

Warnings: 2 unique messages, 2 total
End of script. Logfile hash: a87755456d, CPU: user 2.67s system 0.07s, MEM: 265.95 MB peak
Yosys 0.9+4052 (git sha1 0ccc7229, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)
Time spent: 20% 1x abc9_exe (0 sec), 20% 20x read_verilog (0 sec), ...
Warning: net 'clk_25mhz' does not exist in design, ignoring clock constraint
ERROR: IO 'clock' is unconstrained in LPF (override this error with --lpf-allow-unconstrained)
1 warning, 1 error
sed: can't read build.config: No such file or directory
Failed to open input file

arblake, 2021-04-05:
Rob,
I have reloaded all software and still errors.
First is this:
* boards description file    :  /usr/local/frameworks/boards/boards.json
[ok]
<<=- compiling j1eforth-ULX3S.ice for ulx3s -=>>
using default variant  85k
using variant          85k
using build system     shell
pin set 'us2_ps2' not defined in board variant
make: *** [Makefile:2: ulx3s] Error 255

I removed reference to PS2 from Makefile. As far as I know the board does
not support physically PS2, although it could be implemented.
Next error was this:
=> file:
/home/adrian/Silice-Playground/j1eforth/DE10NANO-ULX3S/ULX3S/../gpu.ice
=> line: 717

simple_dualport_bram_port0 blit1tilemap,
^^^^^^^^^^^^^^^^^^^^^^^^^^ no known interface definition for
'simple_dualport_bram_port0'

error: Silice compiler stopped

I think this is similar to an earlier error.

Adrian

-- 
Adrian Blake, VK2ALF
101 Mulach St
Cooma, NSW, 2630
Australia
Mobile +61 407232978

Pärna tn 3
Otepää
Valgamaa
Estonia
Mobile +372 51971441

arblake, 2021-04-06:
Rob,

It is the WIP that I used , but no success on main branch either.

Adrian

rob-ng15, 2021-04-09:
Hi Adrian,

Just a further thought, have you updated your Silice as j1eforth uses features from the latest WIP branch.

Rob.

arblake, 2021-04-21:
All is working Happy now

sylefeb, 2021-04-21:
Ah that's good to hear! Thanks - closing the issue for now.

closed 2021-04-21 by sylefeb