sylefeb/Silice #135: Verilog error when indexing arrays

Mesabloo, 2021-06-10:
Trying to test this file making use of arrays results in a Verilog error.

<details><summary><b>File to test:</b></summary>

```c
algorithm #main(output uint8 leds) {
  uint8 values[3] = { 0, 1, 2 };

  values[0] = 1;
  values[1] = 2;
  values[2] = 3;

  leds = 1;

//  __display("%d | %d | %d", values[0], values[1], values[2]);
}
```

</details>

<details><summary><b>Verilog error:</b></summary>

- Using Yosys:
  [image]

- Using Icarus:
  [image]

- Using Verilator:
  - Accepts it when the value is never used (most probably optimized away?):
    [image]
  - Errors out when value is used with `__display("%d | %d | %d", values[0], values[1, values[2]);`:
    [image]

</details>

In an attempt to understand why, and probably help fixing this issue, going to the erroring lines yields this Verilog code (reformated):
```verilog
always @(posedge clock) begin
   _q_values[0] <= (reset | ~in_run) ? 0 : _d_values;
   _q_values[1] <= (reset | ~in_run) ? 1 : _d_values;
   _q_values[2] <= (reset | ~in_run) ? 2 : _d_values;
   _q_leds <= _d_leds;
   _q_index <= reset ? 1 : ( ~_autorun ? 0 : _d_index);
   _autorun <= reset ? 0 : 1;
end
```

:point_right: `d_values` is in fact never indexed, which is most probably the issue here. Should it be `_q_values[0] <= (reset | ~in_run) ? 0 : _d_values[0];`?

----------------------------------------

<details><summary><b>Full compilation trace for icarus</b></summary>

```
[nix-shell:~/projects/Silice/draft/tests/issues]$ make -f Makefile.icarus 136_array_discard
silice-make.py -s 136_array_discard.ice -b icarus -p basic -o BUILD_136_array_discard
* Source file                :  /home/mesabloo/projects/Silice/draft/tests/issues/136_array_discard.ice    [ok]
* Silice bin directory       :  /home/mesabloo/projects/Silice/draft/bin
* Build output directory     :  /home/mesabloo/projects/Silice/draft/tests/issues/BUILD_136_array_discard  (exists)
* Silice frameworks directory:  /home/mesabloo/projects/Silice/draft/frameworks 			[ok]
* boards description file    :  /home/mesabloo/projects/Silice/draft/frameworks/boards/boards.json 	[ok]
<<=- compiling 136_array_discard.ice for icarus -=>>
using default variant  configurable
using variant          configurable
using build system     shell
launching command      /home/mesabloo/projects/Silice/draft/frameworks/boards/icarus/icarus.sh /home/mesabloo/projects/Silice/draft/tests/issues/136_array_discard.ice
build script: SILICE_DIR     = /home/mesabloo/projects/Silice/draft/bin
build script: BUILD_DIR      = /home/mesabloo/projects/Silice/draft/tests/issues/BUILD_136_array_discard
build script: BOARD_DIR      = /home/mesabloo/projects/Silice/draft/frameworks/boards/icarus
build script: FRAMEWORKS_DIR = /home/mesabloo/projects/Silice/draft/frameworks
build script: FRAMEWORK_FILE = /home/mesabloo/projects/Silice/draft/frameworks/boards/icarus/icarus.v
rm: cannot remove 'icarus.fst': No such file or directory
rm: cannot remove 'icarus.fst.hier': No such file or directory
preprocessing /home/mesabloo/projects/Silice/draft/frameworks/libraries/memory_ports.ice.
preprocessing /home/mesabloo/projects/Silice/draft/tests/issues/136_array_discard.ice.
                libraries_path = /home/mesabloo/projects/Silice/draft/frameworks/libraries
                framework_file = /home/mesabloo/projects/Silice/draft/frameworks/boards/icarus/icarus.v
              output_fsm_graph =                              1
simple_dualport_bram_wenable1_width =                              1
simple_dualport_bram_wenable1_type =                           uint
                bram_supported =                            yes
                 brom_template =              brom_generic.v.in
                 bram_template =              bram_generic.v.in
                brom_supported =                            yes
  dualport_bram_wenable1_width =                           data
             bram_wenable_type =                           uint
 simple_dualport_bram_template = simple_dualport_bram_generic.v.in
            bram_wenable_width =                           data
       dualport_bram_supported =                            yes
        dualport_bram_template =     dualport_bram_generic.v.in
                templates_path = /home/mesabloo/projects/Silice/draft/frameworks/templates
                frameworks_dir = /home/mesabloo/projects/Silice/draft/frameworks
  dualport_bram_wenable0_width =                           data
   dualport_bram_wenable0_type =                           uint
simple_dualport_bram_supported =                            yes
   dualport_bram_wenable1_type =                           uint
parsing algorithm main
algorithm main num states: 2 (no subs)
---< main::variables >---
values => written but not used
---< main::outputs >---
leds => flip-flop
build.v:237: error: Array _d_values needs an array index here.
build.v:238: error: Array _d_values needs an array index here.
build.v:239: error: Array _d_values needs an array index here.
build.v:172: error: Array _d_values needs an array index here.
build.v:173: error: Array _d_values needs an array index here.
build.v:174: error: Array _d_values needs an array index here.
6 error(s) during elaboration.
build: Unable to open input file.

GTKWave Analyzer v3.3.108 (w)1999-2020 BSI

GTKWAVE | Could not initialize 'icarus.fst', exiting.
```

</details>

**Silice version:** 0.1 - pulled from draft on June 7th

sylefeb, 2021-06-10:
Hi - good catch, this is due to a recent change (draft branch only). Will fix asap!

sylefeb, 2021-06-10:
Fixed! Thanks -

commit b3388a94 (2021-06-10): fixes #135
closed 2021-06-10 by sylefeb