sylefeb/Silice #136: Unable to build on Arch Linux

rob-ng15, 2021-06-11:
`[rob@ArchLinux Silice]$ ./compile_silice_linux.sh ` results in:

`
In file included from /home/rob/Development/github/Silice/src/libs/luabind-deboostified/luabind/make_function.hpp:10,
                 from /home/rob/Development/github/Silice/src/libs/luabind-deboostified/luabind/function.hpp:8,
                 from /home/rob/Development/github/Silice/src/libs/luabind-deboostified/luabind/class.hpp:81,
                 from /home/rob/Development/github/Silice/src/libs/luabind-deboostified/luabind/luabind.hpp:28,
                 from /home/rob/Development/github/Silice/src/libs/luabind-deboostified/src/create_class.cpp:27:
/home/rob/Development/github/Silice/src/libs/luabind-deboostified/luabind/detail/call.hpp: In constructor ‘luabind::detail::invoke_context::invoke_context()’:
/home/rob/Development/github/Silice/src/libs/luabind-deboostified/luabind/detail/call.hpp:47:52: error: ‘numeric_limits’ is not a member of ‘std’
   47 |                                 : best_score((std::numeric_limits<int>::max)())
      |                                                    ^~~~~~~~~~~~~~
`

Using:
`
[rob@ArchLinux Silice]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-werror gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.1.0 (GCC) `
`

sylefeb, 2021-06-11:
Hi Rob - thanks for pointing this out, was it building before on this architecture? The error seems to be in the luabind library which has been there from the start? Might also be a missing `#include <limits>`

rob-ng15, 2021-06-11:
It used to build. I was just trying to update due to changes you have made
and got this error. I think this is the first time I've tried to build
since Arch Linux moved to gcc 11.

Mesabloo, 2021-06-11:
Hi!

This seems strange. Looking through the mentioned code, it's clear why this doesn't compile (there is indeed no `#include <limits>`, even in included headers).
However, what's not clear is why this code compiles(d?) in the first place.
I tried looking for the default included headers, but even Godbolt agrees there using GCC 10.2.0, so `limits` is not one of those.

I unfortunately cannot reproduce this at the moment (as GCC 11 is not yet available on nixpkgs), but I'll be trying to reproduce the issue observed here (it needs time to download an Arch ISO :eyes:).
:point_right: As a side note, everything builds fine with GCC 10.2.0 on NixOS 20.03.3328.6d1a044fc9f (Markhor)

Can you confirm what branch you are pulling from and what was the last time you built the compiler?

rob-ng15, 2021-06-11:
Hi there,

I am using the "draft" branch, as I am using some of the experimental
features. I last built when GCC 10.2.0 was being used by Arch Linux, but it
was recently moved to GCC 11.1.0.

Rob.

Mesabloo, 2021-06-11:
Hi again! :wave: 

After more in-depth research, and spending about half an hour installing Manjaro, I could reproduce the issue with GCC 11.
And...I learned this: https://gitlab.com/wg1/jpeg-xl/-/issues/84
Quoting the interesting part:
> Header dependency changes
> 
> Some C++ Standard Library headers have been changed to no longer include other headers that they do need to depend on. As such, C++ programs that used standard library components without including the right headers will no longer compile.
> The following headers are used less widely in libstdc++ and may need to be included explicitly when compiled with GCC 11:
> ```
> <limits> (for std::numeric_limits)
> <memory> (for std::unique_ptr, std::shared_ptr etc.)
> <utility> (for std::pair, std::tuple_size, std::index_sequence etc.)
> ```

This seems to be the issue we are facing here, where some header included `<limits>` but doesn't anymore.
A simple fix would certainly be to add `#include <limits>` in the following files:
- src/libs/luabind-deboostified/luabind/detail/call.hpp

> Tested on GCC 11.1.0 & GCC 10.2.0; both compiled fine with the header added

sylefeb, 2021-06-11:
Wow, nice catch! merging asap, thanks!

rob-ng15, 2021-06-12:
Working for me, many thanks.

closed 2021-06-12 by rob-ng15