sylefeb/Silice #150: widthof improvement request

rob-ng15, 2021-06-21:
The following circuit would be more flexible with an improvement to widthof(x):

`
circuitry divbit( inout quo, inout rem, input top, input bottom, input x ) {
    sameas( rem ) temp = uninitialized;
    uint1   quobit = uninitialised;

    temp = ( rem << 1 ) + top[x,1];
    quobit = __unsigned(temp) >= __unsigned(bottom);
    rem = __unsigned(temp) - ( quobit ? __unsigned(bottom) : 0 );
    quo[x,1] = quobit;
}
`

Using widthof(x):

`
circuitry divbit( inout quo, inout rem, input top, input bottom, input x ) {
    sameas( rem ) temp = uninitialized;
    uint1   quobit = uninitialised;

    temp = { rem[ 0, widthof(rem)-1, top[x,1] };
    quobit = __unsigned(temp) >= __unsigned(bottom);
    rem = __unsigned(temp) - ( quobit ? __unsigned(bottom) : 0 );
    quo[x,1] = quobit;
}
`

but fails to compile with:

```
temp = { rem[0,widthof(rem)-1, top[x,1] };
                          ^ missing symbol after this

error: [syntax error]
```

sylefeb, 2021-07-15:
Hi Rob, that is a very good point ; requires const expression collapsing which is on my todo (but I don't think I had opened an issue for it yet). Thanks!

rob-ng15, 2021-09-07:
Further potential example:

// ABSOLUTE VALUE
circuitry abs( input value1, output  absolute ) {
    absolute = ( __signed(value1) < __signed(0) ) ? -value1 : value1;
}

could be:

circuitry abs( input value 1, output absolute ) {
  absolute = value1[ widthof(value1)-1,1 ] ? -value1 : value1;
}

which would save a comparison.