sylefeb/Silice #152: Variable mistakenly generated as flip-flop where it was only temporary

Mesabloo, 2021-06-28:
Struggled quite a bit to find a minimal example from a 300-ish lines long code, but here it goes:
```c
algorithm main(output uint8 leds) {
  subroutine f(
    input uint1 bool
  ) {
    uint1 x <: bool;
  }

  () <- f <- (0);
}

```

This example generates incorrect Verilog code, assuming `bool` (in the subroutine `f`) is a flip-flop, where in fact it is a temporary variable.
This leads to errors such as:
[image]

I hope this one has not yet been reported!

sylefeb, 2021-06-29:
The problem runs a bit deeper than it seems, with dependencies not properly propagated through expression trackers (<:). Fix is in progress.

sylefeb, 2021-06-29:
Fixed!

closed 2021-06-29 by sylefeb