sylefeb/Silice #162: How can I get only the verilog code?

LightCannon, 2021-07-13:
Hello. 
Thanks for this promising projects.
I want to get the verilog code only from the .ice file (without flashing to any board,..etc). Just the verilog RTL code. I cannot find anywhere in the docs a mention for that thing. May I know if this is supported? and how to do it? 
Thanks in advance

sylefeb, 2021-07-14:
Hi - thanks for reaching out! This is possible, but could be made simpler (so let's leave this issue open).
I am using https://github.com/sylefeb/Silice/tree/draft/projects/ice-v as the example.

If you look at the project Makefile you will see something like this:
```make
.DEFAULT: ice-v.ice
		silice-make.py -s ice-v-soc.ice -b $@ -p basic -o BUILD_$(subst :,_,$@)
```
The makefile is invoking the silice build system, which is the `silice-make.py` script.

This can be called from the command line directly, adding the options `--no_build` and `--no_program`, for instance
```shell
  silice-make.py -s ice-v-soc.ice -b icestick -p basic -o BUILD_icestick --no_build --no_program
```
This will only generate the verilog output `build.v` in `BUILD_icestick` (compiled for the icestick board).

LightCannon, 2021-07-14:
Thanks,
Is there a way I can get only the code,without all the added FPGA related logic?  (for example, Blinky generated code is too long because it is adding some parts related to the FPGA itself, I just need the main core of the logic written, like module M_main in the generated code for blinky).

sylefeb, 2021-07-15:
Hi - yes this is what I'd like to facilitate.

One simple way would be to create an 'empty' board (https://github.com/sylefeb/Silice/tree/master/frameworks/boards) ; but any used default var (e.g. `NUM_LEDS`) would have to be defined in the source. 

I'll have a look asap --

LightCannon, 2021-07-15:
Thanks, I will try that. 
Looking forward to seeing this feature integrated transparently in Silice.

sylefeb, 2021-07-15:
I've just pushed on draft a 'bare' board that should help. Simply use `make bare`.

LightCannon, 2021-07-16:
I have tried using "bare" board then compiling in Modelsim. 
I got errors on all ports "Net type of  <port name> must be explicitly declared". This is due to the generated Verilog has the ports not using the default_nettype define. 
I wonder if this is a bug or there is something I missed?
I ran this directly 
`
silice -f F:\Silice\Silice\frameworks\boards\bare\bare.v F:/Silice/Silice/projects/blinky/blinky.ice -o build_bare.v
`

sylefeb, 2021-07-18:
Hi - command line looks good ; I just tried to launch `verilator --lint-only build_bare.v` on it, no issues are reported. I never used Modelsim ; does it require a testbench? Then the closest 'board' would [`icarus`](https://github.com/sylefeb/Silice/tree/master/frameworks/boards/icarus) which adds a testbench to the design. 

The `bare` framework specifies `default_nettype none` such that default declarations are not allowed, but the generated code should be ok in this respect. There is however no top module or testbench for simulation, only the bare compiled code.

(Would be good to add a Modelsim target though ... installing!)