sylefeb/Silice #167: Taking width of circuitry VIO

Mesabloo, 2021-08-02:
At the moment, trying to call `widthof` on a `circuitry` VIO results in an error, as can be seen in this MWE:
```c
circuitry take_width(input dat, output width) {
                    //     ^^^ Be careful: this must not reference a variable in the scope
                    //         where `take_width` is instantiated!
  width = widthof(dat);
}

algorithm main(output uint$NUM_LEDS$ leds) {
  uint20 data  = uninitialized;
  uint6  width = uninitialized;

  (width) = take_width(data);

  __display("WIDTH: %d", width);
}
```
[image]

rob-ng15, 2021-08-02:
Hi there,

I think this is a duplicate of https://github.com/sylefeb/Silice/issues/150
?

sylefeb, 2021-08-02:
Indeed, even though #150 also requires an additional step to allow const expressions to be resolved at compile time. I'll try to tackle both at once!