sylefeb/Silice #170: hdmi.ice and subcomponents lead to unstable display

rob-ng15, 2021-08-20:
Replacing an older hdmi.ice ddr.v hdmi_clock.v and hdmi_ddr_crgb.v with the latest from the draft branch of Silice causes PAWS to have an unstable HDMI output, screen flashing on and  off.

Will have a look at the kdiff between the files and see if I can track down the issue further.

rob-ng15, 2021-11-04:
> Hi Sylvain,
>
> I'll try again with the update versions from your repository and see what
> breaks!
>
> Rob.

Just rebuilt PAWS with the updated hdmi.ice and hdmi_clock.v from Silice
draft. Had to change both as changing just hdmi.ice failed to build due to
a change in hdmi_clock.v

Display loses sync, via the capture device you get constant flashes of
picture and sync loss. Via monitor, get out of sync message.

The diffs from hte version I am presently using to the latest in Silice
DRAFT branch.

@@ -11,15 +11,12 @@
// - https://www.fpga4fun.com/HDMI.html
// -
https://github.com/lawrie/ulx3s_examples/blob/master/hdmi/tmds_encoder.v
//
//      GNU AFFERO GENERAL PUBLIC LICENSE
//        Version 3, 19 November 2007
//
//  A copy of the license full text is included in
//  the distribution, please refer to it for details.
// MIT license, see LICENSE_MIT in Silice repo root

import('hdmi_clock.v')
import('ddr.v')
import('hdmi_ddr_crgb.v')
$include('clean_reset.ice')

// ----------------------------------------------------

@ -34,29 +31,29 @@ algorithm tmds_encoder(
  int5  dc_bias         = uninitialized;

  // tracks 'number on ones' in input
  uint4 num_ones        := data[0,1] + data[1,1] + data[2,1] + data[3,1]
  uint4 num_ones        <: data[0,1] + data[1,1] + data[2,1] + data[3,1]
                         + data[4,1] + data[5,1] + data[6,1] + data[7,1];
  // tracks 'numbers of ones minus number of zeros' in internal byte
  int5  diff_ones_zeros := q_m[0,1] + q_m[1,1] + q_m[2,1] + q_m[3,1]
                         + q_m[4,1] + q_m[5,1] + q_m[6,1] + q_m[7,1] - 6d4;
  int5  diff_ones_zeros <:: q_m[0,1] + q_m[1,1] + q_m[2,1] + q_m[3,1]
                          + q_m[4,1] + q_m[5,1] + q_m[6,1] + q_m[7,1] - 6d4;

  // XOR chain on input
  int1  xored1          := data[1,1] ^ data[0,1];
  int1  xored2          := data[2,1] ^ xored1;
  int1  xored3          := data[3,1] ^ xored2;
  int1  xored4          := data[4,1] ^ xored3;
  int1  xored5          := data[5,1] ^ xored4;
  int1  xored6          := data[6,1] ^ xored5;
  int1  xored7          := data[7,1] ^ xored6;
  int1  xored1          <: data[1,1] ^ data[0,1];
  int1  xored2          <: data[2,1] ^ xored1;
  int1  xored3          <: data[3,1] ^ xored2;
  int1  xored4          <: data[4,1] ^ xored3;
  int1  xored5          <: data[5,1] ^ xored4;
  int1  xored6          <: data[6,1] ^ xored5;
  int1  xored7          <: data[7,1] ^ xored6;

  // XNOR chain on input
  int1  xnored1         := ~(data[1,1] ^ data[0,1]);
  int1  xnored2         := ~(data[2,1] ^ xnored1);
  int1  xnored3         := ~(data[3,1] ^ xnored2);
  int1  xnored4         := ~(data[4,1] ^ xnored3);
  int1  xnored5         := ~(data[5,1] ^ xnored4);
  int1  xnored6         := ~(data[6,1] ^ xnored5);
  int1  xnored7         := ~(data[7,1] ^ xnored6);
  int1  xnored1         <: ~(data[1,1] ^ data[0,1]);
  int1  xnored2         <: ~(data[2,1] ^ xnored1);
  int1  xnored3         <: ~(data[3,1] ^ xnored2);
  int1  xnored4         <: ~(data[4,1] ^ xnored3);
  int1  xnored5         <: ~(data[5,1] ^ xnored4);
  int1  xnored6         <: ~(data[6,1] ^ xnored5);
  int1  xnored7         <: ~(data[7,1] ^ xnored6);

  always {
    // choice of encoding scheme (xor / xnor)
@ -129,25 +126,32 @@ algorithm hdmi(
  output  uint10 y,
  output  uint1  active,
  output  uint1  vblank,
  output! uint4  gpdi_dp,
  output! uint4  gpdi_dp,
  input   uint8  red,
  input   uint8  green,
  input   uint8  blue,
)  {
) ***@***.***_clk,!rst> {

  uint10 cntx  = 0;
  uint9  cnty  = 0;

  uint1 hsync      := (cntx > 655) && (cntx < 752);
  uint1 vsync      := (cnty > 489) && (cnty < 492);
  uint1 hsync      <: (cntx > 655) && (cntx < 752);
  uint1 vsync      <: (cnty > 489) && (cnty < 492);

  uint2 sync_ctrl   = uninitialized;
  uint2 null_ctrl  := 0;
  uint2 null_ctrl  <: 0;

  // pll for tmds
  uint1  rst  = uninitialized;
  clean_reset ***@***.***_clk,!reset>(
    out :> rst
  );

  // pll for tmds
  uint1  pixel_clk     = uninitialized;
  uint1  half_hdmi_clk = uninitialized;
  hdmi_clock pll(
    clk           <: clock,         //  25 MHz
    clk           <: clock,
    pixel_clk     :> pixel_clk,     //  25 MHz
    half_hdmi_clk :> half_hdmi_clk, // 125 MHz (half 250MHz HDMI, double
data rate output)
  );

@ -185,9 +189,9 @@ algorithm hdmi(
  // shifter
  uint8 crgb_pos = 0;
  hdmi_ddr_shifter ***@***.***_hdmi_clk>(
    data_r  <: tmds_red,
    data_g  <: tmds_green,
    data_b  <: tmds_blue,
    data_r    <: tmds_red,
    data_g    <: tmds_green,
    data_b    <: tmds_blue,
    p_outbits :> crgb_pos
  );

@ @@

@@ -1,14 +1,26 @@
module hdmi_clock (
        input  clk,           //  25 MHz
        input  clk,
        input  rst,
        output pixel_clk,     // 25 MHz
        output half_hdmi_clk  // 125 MHz
    );

`ifdef MOJO

mojo_clk_50_25_125_125n mclk(
  .CLK_IN1(clk),
  .CLK_OUT2(pixel_clk),
  .CLK_OUT3(half_hdmi_clk),
  .CLK_OUT4(half_hdmi_clk_n),
  .RESET(rst)
);

`else

`ifdef DE10NANO

// not yet implemented

`else

`ifdef ULX3S
@ -59,6 +71,8 @@ EHXPLLL #(
        .LOCK(locked)
);

assign pixel_clk = clk;

`else

`ifdef ICARUS
@ -71,6 +85,7 @@ initial begin
end

assign half_hdmi_clk = genclk;
assign pixel_clk     = clk;

`endif

@ @@

sylefeb, 2021-11-09:
Thanks! I think I broke it when doing the mojov3 version. I'll start looking into this.