sylefeb/Silice #171: Segfault with pipelines

rob-ng15, 2021-09-02:
[VERILATOR.zip](https://github.com/sylefeb/Silice/files/7097672/VERILATOR.zip)

These two files for VERILATOR should give the same output. pipeline.ice gives a segfault, it attempts to use a pipeline to render the pixels for the circle. circle.ice gives the correct result (the original code from PAWS), pipeline.ice is an experiment.

sylefeb, 2021-09-03:
The code has a syntax error (the error reporting was broken, hence the crash, fixed!): the pipeline is outputting from multiple stages to outputs `bitmap_y_write` and `bitmap_x_write`. This is not possible as all stages are active simultaneously, so only one can write the output at a given cycle. Makes sense?

A typical pipeline pattern would be to pass a value from one stage to the next, as `to_insert` in the pipelines sort example: it is written at stage 0, then read and written at every subsequent stages. Each stage in fact has its own copy of `to_insert` and the value 'trickle' down the pipeline automatically.

rob-ng15, 2021-09-03:
OK, yes it was a misunderstanding of how the pipelines worked on my part, I
thought they would follow one stage to the next.

Rob.

commit ea550553 (2021-09-03): fixes #171, missing argument in error reporting
closed 2021-09-03 by rob-ng15