sylefeb/Silice #174: Getting started/macOS: no default-jre

tommythorn, 2021-09-04:
Kudos on an exceptionally well documented project.

Alas, trying to follow the macOS setup instructions didn't succeed for me as "default-jre" doesn't appear to be a valid target on my (fully up-to-date) brew installation on macOS Mojave (deliberately held back):
```
$ brew install default-jre default-jdk git uuid ossp-uuid
Warning: No available formula or cask with the name "default-jre". Did you mean defaultbrowser?
==> Searching for similarly named formulae...
This similarly named formula was found:
defaultbrowser
To install it, run:
  brew install defaultbrowser
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

sylefeb, 2021-09-04:
Hi - thanks! I unfortunately have zero experience with macOS, but let's try to get this to work together! Also maybe @osnr and @ThibaultTricard can help has I think they got Silice to compile on macOS.

These jre/jdk targets are Java, which is required during the compilation of Silice (and only then). That is because Silice relies on ANTLR, a Java library that produces C++ code for parsing files. So I would assume the packages exist but have changed names?

ThibaultTricard, 2021-09-04:
Hi,

I got Silice to compile on mac to compile on mac, but that was a few month ago now
so I don't remember exactly all the steps : for java I think there is two option : 

- brew install java which install the openjdk
- or a manual install using a .dmg

I honestly don't remember which one I did

If I remember correctly the tricky part is the getting the toolchain.
From the note I Took back then : 
<<
  ```
  brew tap ktemkin/oss-fpga
  brew install --HEAD icestorm yosys nextpnr-ice40 project-trellis nextpnr-trellis verilator icarus-verilog 
  ```
  
  To install gtkwave checkout this git [gtkwave](https://ughe.github.io/2018/11/06/gtkwave-osx) for detailled instruction
>>

tommythorn, 2021-09-19:
Java 17 was released recently and just running `javac` on macOS offers to take you to a page where you can install it.  So that's probably the most official path.  I was able to install on macOS 11.6 (BigSur), but alas not on macOS *10.14.6* (Mojave, where 32-bit binaries still roam freely).

I don't expect too many will care about my woes, but just in case, these are the build issues I hit:
```
Consolidate compiler generated dependencies of target silice
[ 55%] Building CXX object CMakeFiles/silice.dir/src/LuaPreProcessor.cpp.o
/tmp/tank/projects/Silice/src/LuaPreProcessor.cpp:622:33: error: 'absolute' is unavailable: introduced in macOS 10.15
    freport << std::filesystem::absolute(src_file).string() << '\n';
                                ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:1477:39: note: 'absolute' has been explicitly
      marked unavailable here
inline _LIBCPP_INLINE_VISIBILITY path absolute(const path& __p) {
                                      ^
/tmp/tank/projects/Silice/src/LuaPreProcessor.cpp:622:42: error: 'path' is unavailable: introduced in macOS 10.15
    freport << std::filesystem::absolute(src_file).string() << '\n';
                                         ^
...
```

closed 2021-10-17 by tommythorn