sylefeb/Silice #175: silice-make.py doesn't launch commands

GuzTech, 2021-09-06:
On my system (RebornOS, an Arch Linux derivative), silice-make.py doesn't launch any command. I use ZSH, but I've also tried it with bash without success. I've 'fixed' it by removing the `bash + " " +` part from this line: https://github.com/sylefeb/Silice/blob/master/bin/silice-make.py#L232

The `bash` variable contains  the string`env bash`, which doesn't do anything on my system. So simply changing this:

```python
os.system(bash + " " + command + " " + defines)
```
to this:
```python
os.system(command + " " + defines)
```
fixes the problem for me.

sylefeb, 2021-09-06:
Thanks for the detailed report, I'll look into it asap!

sylefeb, 2021-09-06:
Should be fixed now (draft branch), simply made sure the 'env bash' part is only called if on Windows/MinGW. Hope this does not affect macOS users though ... (@ThibaultTricard could you give it a quick test? thanks!).

ThibaultTricard, 2021-09-06:
I quicly recompiled Silice on this branch (did a pull request on some small compile issues),
I tried the multiplication project
make verilator doesn't works but make icarus work
probably my verilator tool chain is a bit old

GuzTech, 2021-09-06:
I have tried Icarus and Verilator (both the latest version) with the divstd_bare and doomchip demos, and they work fine without the `env bash` part on my system.

sylefeb, 2021-09-07:
Excellent - I'll close the issue. @ThibaultTricard there is a known issue with Verilator where prior versions do not know the `-Wno-TIMESCALEMOD` parameter and fail. I'll open a separate issue.