sylefeb/Silice #186: compile error in ubuntu 18.04(wsl)

niyazFattahov, 2021-09-17:
I follow this: https://github.com/sylefeb/Silice/blob/master/GetStarted_Linux.md
when compiling I got:
LuaPreProcessor.cpp:38:10: fatal error: filesystem: No such file or directory
 #include <filesystem>
I added "experimental" after googling.
Then I get:

[ 94%] Linking CXX executable silice
src/libs/lua/liblua.a(loslib.c.o): In function `os_tmpname':
loslib.c:(.text+0x2b0): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
CMakeFiles/silice.dir/src/LuaPreProcessor.cpp.o: In function `LuaPreProcessor::processCode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >)':
LuaPreProcessor.cpp:(.text+0x7698): undefined reference to `std::experimental::filesystem::v1::current_path[abi:cxx11]()'
LuaPreProcessor.cpp:(.text+0x76ee): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
LuaPreProcessor.cpp:(.text+0x7708): undefined reference to `std::experimental::filesystem::v1::absolute(std::experimental::filesystem::v1::__cxx11::path const&, std::experimental::filesystem::v1::__cxx11::path const&)'
LuaPreProcessor.cpp:(.text+0x7872): undefined reference to `std::experimental::filesystem::v1::current_path[abi:cxx11]()'
LuaPreProcessor.cpp:(.text+0x78c8): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
LuaPreProcessor.cpp:(.text+0x78e0): undefined reference to `std::experimental::filesystem::v1::absolute(std::experimental::filesystem::v1::__cxx11::path const&, std::experimental::filesystem::v1::__cxx11::path const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/silice.dir/build.make:682: recipe for target 'silice' failed
make[2]: *** [silice] Error 1
CMakeFiles/Makefile2:69: recipe for target 'CMakeFiles/silice.dir/all' failed
make[1]: *** [CMakeFiles/silice.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

Could you please help me?

sylefeb, 2021-09-17:
Hi - thanks for trying Silice! This is most likely due to the version of gcc being a bit old (7.4-ish?). I did compile on Ubuntu 18.04 installing a more recent gcc, that is fortunately possible, see e.g. https://linuxize.com/post/how-to-install-gcc-compiler-on-ubuntu-18-04/#installing-multiple-gcc-versions

I had never tried with `#include <experimental/filesystem>` -- it is unfortunate that does not work ; the best is likely to upgrade gcc, hoping that is possible on your system?

niyazFattahov, 2021-09-17:
thanks for help, I installed gcc-9 and no compile error

closed 2021-09-17 by niyazFattahov