sylefeb/Silice #203: Linux builds on `draft` broken due to permissions change

tommythorn, 2022-01-15:
I was trying out the latest `draft` branch, but the build failed with
```
...
/bin/sh: 1: /home/tommy/projects/Silice/antlr/compile.sh: Permission denied
/bin/sh: 1: /home/tommy/projects/Silice/antlr/compile.sh: Permission denied
/bin/sh: 1: make[2]: *** [CMakeFiles/silice.dir/build.make:94: ../../antlr/vmoduleParser.cpp] Error 126
/home/tommy/projects/Silice/antlr/compile.sh: Permission denied
...
```
I bisected it down to change 89e1190a0505ac440ad43aede1fcdea119bb4eb8 which indeed blows away the permission on a lot of files.

sylefeb, 2022-01-15:
Thanks for this, I'll investigate the root cause, keeping open for now.

sylefeb, 2022-01-15:
I have identified the source of the problem (one of my repo clone was improperly setup). I have fixed a few additional permissions and re-tested Linux builds on master/wip/draft. Everything seems in order, so closing. Thanks again!

commit 1acb7ee8 (2022-01-15): Restore executable permissions on antlr scripts; Fixes #203
commit dd551971 (2022-01-15): Merge pull request #205 from tommythorn/fix-203

Restore executable permissions on antlr scripts; Fixes #203
closed 2022-01-15 by sylefeb