sylefeb/Silice #206: File extension

zipotron, 2022-01-17:
Hello, first thing congratulate for this awesome work, I just discover this project last weekend and got in love with it.
I am writing this issue because, I discover by accident Silice. I found an interesting work (The PAWSv2 risc-v softcore) and I wanted to check it, when I saw the .ice extension I recognize it as a IceStusio files, and then try to open with my IceStudio, other amazing opensource project.
The big surprise was to discover that those .ice files was corresponding to another completely different project, this one, and the surprise became even bigger when discover that Github is not making the difference between the IceStudio code and the Silice code (for Github everything is "Slice") .
BTW Github is tagging .ice as a "Slice" not "Silice".

sylefeb, 2022-01-17:
Hi - thanks and welcome to Silice! PAWS is great, so many excellent features and games!

Extension: that is a good point. Unfortunately I wasn't aware about the file extension collision until late in the project, and I agree this is not ideal. I've since been thinking of changing it. This has little technical impact but a rather wide one in terms of repository. I'll bump this up in priority.

rob-ng15, 2022-01-17:
Cheers for looking at PAWSv2, I hope you like it.

Silice is a great environment for learning about FPGA programming, and
sylefeb is great as a project leader on Silice.

sylefeb, 2022-01-17:
Hi Rob, thanks :-) Cheers!

sylefeb, 2022-01-17:
A few ideas on extensions:
- ".silice", after all most systems should support more than 3 letters? (e.g. ".html")
- ".si" short, does not seem too crowded, at least not in the FPGA space. Also, that's Silicon in the element table.
- ".ic" is not bad either, and has the advantage to be a single letter change.

rob-ng15, 2022-01-17:
I'm happy with any of those, but I do like .silice as definitive, exactly
as per tin (don't worry if you don't get this reference, English joke).

umarcor, 2022-01-17:
@zipotron, I think that GitHub's `Slice` refers to a third file format. It's Slice (Specification Language for Ice) and Internet Communications Engine (Ice):

- https://github.com/github/linguist/blob/80f3531e8a1014a23f4606458e5a528053ed3cac/lib/linguist/languages.yml#L5730-L5737
- https://doc.zeroc.com/ice/3.6/introduction

The extension is also listed in https://github.com/github/linguist/blob/80f3531e8a1014a23f4606458e5a528053ed3cac/lib/linguist/languages.yml#L2707, which is the Icestudio case I believe.

> A few ideas on extensions:
>
> * ".silice", after all most systems should support more than 3 letters? (e.g. ".html")
> * ".si" short, does not seem too crowded, at least not in the FPGA space. Also, [...]

IIRC, there are historical reasons to use three characters for extensions. That might be irrelevant nowadays, but worth giving a thought. In the VHDL community, there is fragmentation between `.vhd` and `.vhdl` for that reason. See the TIP in http://ghdl.github.io/ghdl/quick_start/simulation/hello/index.html.

In this case, the content of Silice's `.ice` files should be very obviously different from Icestudio's (which are JSON) for any parser/analyser. "*"If it walks like a duck and it quacks like a duck, then it must be a duck"*". Therefore, it might be reasonable to acknowledge the polymorphism rather that try changing the user bases.

Other than that, I like that `.si` is the symbol in the element table.

zipotron, 2022-01-17:
Thanks Unai! Well, And .sil? Or .slc (just consonants)?

zipotron, 2022-01-17:
Just for fill the three characters, because the idea of .si is cool

sylefeb, 2022-01-17:
I thought of `.slc` as well but this one is used for slice formats in Additive Manufacturing (since there is some confusion between Silice and Slice it might reinforce it further, plus I also do AM software so I might get confused myself ;) )
`.sil` is definitely an option, but I like `.si` more due to the element table.  Also it seems <=3 character should be fine? (I am thinking of python's `.py` for instance)

umarcor, 2022-01-17:
I think that two characters (`.si`) is ok. As @sylefeb said, several well-known extensions have one or two characters. The limitation in "8.3 filename" is as follows:

https://en.wikipedia.org/wiki/8.3_filename
> *8.3 filenames are limited to at most eight characters (after any directory specifier), followed optionally by a filename extension consisting of a period . and at most three further characters. For systems that only support 8.3 filenames, excess characters are ignored.*

zipotron, 2022-01-17:
I definitely like .si

sylefeb, 2022-04-08:
Closing ; draft branch has now transitioned to `.si` (merge with master planned soon). Thanks everyone for the discussion!

closed 2022-04-08 by sylefeb