sylefeb/Silice #207: ULX3S reboot issue

nono2357, 2022-01-31:
ULX3S boards reboot after a few seconds after being programmed because their `wifi_gpio0` pin should be kept high.

In `frameworks/boards/ulx3s/ulx3s.v`, those 2 lines should be added:
```
output wifi_gpio0
(...)
assign wifi_gpio0 = 1'b1;
```

Thanks!

sylefeb, 2022-01-31:
Hi - thanks for the report! I'lI add the lines to the framework. 

What surprises me is that I have been using several (quite a few in fact ;) ) ULX3S boards and never got this problem so far. Do you have a particular setup? Maybe something to do with the configuration of the ESP32? Would be nice to understand the root cause! (@rob-ng15 did you encounter this?)

nono2357, 2022-01-31:
Thank you for your understanding. I have an official ULX3S v3.0.3 ordered from Radiona, with the onboard ESP32. I have never used the ESP32 but this fix is needed and well known in order to avoid reboots, see for example:
- https://github.com/emard/ulx3s-misc/blob/master/examples/dvi/top/top_vgatest.v : "wifi_gpio0=1 keeps board from rebooting"
- https://github.com/mit-plv/koika/blob/master/examples/rv/etc/top_ulx3s.v : "Tye GPIO0 to prevent ULX3S from rebooting"

rob-ng15, 2022-01-31:
Hi there,

I've looked through everything I've done and never set this flag. No issues
with rebooting.

emard, 2022-01-31:
This "reboot" is "normal" for boards shipped with old esp32
firmware written in C/arduino
i
t uses ssd1331 display and gpio0 state to take control of the
board. Although intended to be "user friendly", it is not for
us, we are power users and we are more comfortable with
some command prompt from esp32 so

I suggest to flash esp32 with micropython 1.14 and install
latest esp32ecp5. gpio0 will not be checked and board will
not "reboot" if gpio0 is not held 1 by new bistream

see details with many different ways how to install esp32ecp5
https://github.com/emard/esp32ecp5
(install over usb, over internet, different ways)

Send issue if something is not working as it should

sylefeb, 2022-02-01:
Thanks emard for the additional info! To be sure I've added the suggested fix (with ref to this issue), so that the problem should not occur again.

emard, 2022-02-01:
Yes, it is all ok with the core,

with same core board can behave different when firmware
in esp32 is different, micropython is neutral
and easy to control

sylefeb, 2022-02-28:
Closing as the framework now includes the required line, draft branch: https://github.com/sylefeb/Silice/blob/819e81b05786dad69822f89ee766f421b00067aa/frameworks/boards/ulx3s/ulx3s.v#L342

Thanks!

closed 2022-02-28 by sylefeb