sylefeb/Silice #210: Enhancement Request: Import 16 bit binary files

rob-ng15, 2022-02-26:
When including a file that is 16 bit it has to be included as:

    // RISC-V FAST BRAM and BIOS
    bram uint16 ram[16384] = {
        $include('ROM/BIOS.inc')
        , pad(uninitialized)
    };

with 16bit hex entries.

Binary imports are allowed for 8 bit and 32 bit, can't remember the mechanism off the top of my head, but is it possible to extend this to 16 bit?

rob-ng15, 2022-02-26:
> Hi Rob, this is a good point, I'll look into it! Syntax to load:
> https://github.com/sylefeb/Silice/blob/67e99e3f9b7fa0f4924da729ba222d95ed574cec/learn-silice/Documentation.md?plain=1#L219-L223

It'll just make things easier for me, I can just point directly to the
compiled BIOS libraries, rather than have to convert them.

sylefeb, 2022-02-28:
Hi Rob, from the source code I just realized uint16/int16 are already supported! Sorry I forgot to document it, fixing.

rob-ng15, 2022-03-01:
Closing as working. Thank you.

closed 2022-03-01 by rob-ng15