sylefeb/Silice #212: build.v: No such file or directory?

ShervinShokouhi, 2022-03-14:
I followed "Getting started with Silice" and "Getting started under Windows" for WIN 10 64bit, Any help will be appreciated

[image]
.

rob-ng15, 2022-03-14:
Do you have Icarus installed?

rob-ng15, 2022-03-14:
Apologies, line 34 of the icarus.sh file is calling silice. It looks like
it isn't finding silice in your path. Check that the directory where the
silice executable resides is in your path?

sylefeb, 2022-03-14:
Thanks for trying Silice @ShervinShokouhi! @rob-ng15 is correct, either the path is not setup or the executable is not there.

Under mingw64 the `get_started_mingw64.sh` script is supposed to have set up the path in your `.bashrc`:

https://github.com/sylefeb/Silice/blob/31fe0c204d7d7b3f5a5190d4e3944211fc68da6b/get_started_mingw64.sh#L26

You can do that manually in the mingw64 shell by entering `export PATH=$PATH:'$DIR/bin':'$DIR/tools/fpga-binutils/mingw64/bin`, replacing `$DIR` by your Silice root directory (the one where you cloned the repo).

After making sure the `silice.exe` is in `$DIR/bin`, could you also check the end of your ~/.bashrc file to see whether the line was added?

ShervinShokouhi, 2022-03-14:
"After making sure the silice.exe is in $DIR/bin?"
Currently there isn't any "silice.exe" in "bin" directory, Is it created during "get_started_mingw64.sh" or some commands must be run manually?
And also how to be sure all steps during "get_started_mingw64.sh" are executed correctly?

sylefeb, 2022-03-14:
Yes it should have been created. The `get_started_mingw64.sh` script calls the  `compile_silice_mingw64.sh` script. Try to launch ` ./compile_silice_mingw64.sh` directly and send the log (console output) here if that fails? 
Be sure to be in a MinGW**64** shell, in the same directory as the script. The prompt should look like that:  
[image]

ShervinShokouhi, 2022-03-14:
inflating: jdk-14.0.1/release
mkdir: cannot create directory ‘build-silice’: File exists
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at antlr/antlr4-cpp-runtime-4.7.2-source/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Building without demo. To enable demo build use: -DWITH_DEMO=True
CMake Deprecation Warning at antlr/antlr4-cpp-runtime-4.7.2-source/CMakeLists.txt:31 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at antlr/antlr4-cpp-runtime-4.7.2-source/CMakeLists.txt:32 (CMAKE_POLICY):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at antlr/antlr4-cpp-runtime-4.7.2-source/CMakeLists.txt:33 (CMAKE_POLICY):
  The OLD behavior for policy CMP0042 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at antlr/antlr4-cpp-runtime-4.7.2-source/CMakeLists.txt:38 (CMAKE_POLICY):
  The OLD behavior for policy CMP0059 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at antlr/antlr4-cpp-runtime-4.7.2-source/CMakeLists.txt:39 (CMAKE_POLICY):
  The OLD behavior for policy CMP0054 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Output libraries to D:/Silice/Silice/lib
CMake Deprecation Warning at src/libs/lua/CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/libs/luabind-deboostified/CMakeLists.txt:6 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
CMake Error at CMakeLists.txt:70 (ADD_EXECUTABLE):
  Cannot find source file:

    src/libs/LibSL-small/src/LibSL/System/System.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at CMakeLists.txt:70 (ADD_EXECUTABLE):
  No SOURCES given to target: silice

CMake Generate step failed.  Build files cannot be regenerated correctly.
mingw32-make: *** No rule to make target 'install'.  Stop.

--------------------------------------------------------------------
Please start a new shell before using Silice (PATH has been changed)
--------------------------------------------------------------------

ShervinShokouhi, 2022-03-14:
Thanks @sylefeb:
Yes "src/libs/LibSL-small" dir was empty!
First fix it with: "git submodule update --init --recursive"
Then: "./compile_silice_mingw64.sh", It make "Silice.exe" in "bin"
And again: "./compile_silice_mingw64.sh"
Last: Test "divstd_bare" by icarus lead to "icarus.fst"

sylefeb, 2022-03-14:
So all seems to be working now? A good test is to run `./simul.sh` from within `projects/terrain` (needs verilator, but that should be in the downloaded package already).

ShervinShokouhi, 2022-03-15:
Fantastic, It works like a charm
Thank you

closed 2022-03-15 by ShervinShokouhi

sylefeb, 2022-03-15:
That's great! Hope you'll enjoy Silice!

ShervinShokouhi, 2022-03-15:
For sure I will be ;)