sylefeb/Silice #218: Difficulties to download Silice.

at91rm9200, 2022-05-07:
Hello Sylvain,

I am trying to download Silice with

   git clone --recurse-submodules https://github.com/sylefeb/Silice.git

but eventually I get the following error message:   
...
remote: Enumerating objects: 40, done.        
remote: Total 40 (delta 0), reused 0 (delta 0), pack-reused 40        
Cloning into '/home/bernd/build/Silice/004/Silice/python/pybind11'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Are there further steps required to download Silice?  I hope I don't have to mess around with ssh keys.

Regards, Bernd.

sylefeb, 2022-05-08:
Hello Bernd,

Thank you very much for pointing this out, I have fixed the issue which was due to using a ssh url when I added pybind11 as a submodule. I had not noticed due to my keys always being available.

This should be fixed, could you please try again?

Best regards,
Sylvain

at91rm9200, 2022-05-08:
Hello Sylvain,

thank you. It is fixed now.

Regards, Bernd.

closed 2022-05-08 by at91rm9200