sylefeb/Silice #219: How to use newlib to compile c code?

ShervinShokouhi, 2022-05-15:
Assume you have below code (neglect about puts and putchar) how I cane use newlib to compile this code?

Compile cmd:
riscv32-unknown-elf-gcc -fno-unroll-loops -O1 -fno-stack-protector -fno-pic -march=rv32i -mabi=ilp32 -c -o test.o test.c

Code:
#include <stdio.h>
int main()
{
  int k = 2;
  float i,j,r,x,y=-16;
  while (puts(""), y++<15)
    for (x=0; x++<84; putchar(" .:-;!/>)|&IH%*#"[k&15]))
      for (i=k=r=0; 
           j=r*r-i*i-2+x/25,i=2*r*i+y/10,j*j+i*i<11&&k++<111; 
           r=j);
  return 0;
}

I get these error messages:
 riscv32-unknown-elf-ld  -m elf32lriscv -b elf32-littleriscv --no-relax  -lm -o test.elf test.o               F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: warning: cannot find entry symbol _start; defaulting to 0000000000010074
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.o: in function `main':
test.c:(.text+0x40): undefined reference to `puts'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0xb4): undefined reference to `putchar'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0xc8): undefined reference to `__addsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0xdc): undefined reference to `__ltsf2'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0xf4): undefined reference to `__divsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x108): undefined reference to `__divsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x130): undefined reference to `__mulsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x144): undefined reference to `__mulsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x154): undefined reference to `__subsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x160): undefined reference to `__subsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x16c): undefined reference to `__addsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x180): undefined reference to `__addsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x18c): undefined reference to `__mulsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x198): undefined reference to `__addsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x1ac): undefined reference to `__mulsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x1c0): undefined reference to `__mulsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x1d0): undefined reference to `__addsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x1dc): undefined reference to `__ltsf2'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x1fc): undefined reference to `puts'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x20c): undefined reference to `__addsf3'
F:\Silice\Toolchains\riscv32-unknown-elf\bin\riscv32-unknown-elf-ld.exe: test.c:(.text+0x220): undefined reference to `__ltsf2'

Help me about GCC, AS, LD and rest settings to create HEX file from C with NewLib support.

rob-ng15, 2022-05-16:
You'll need to pass a flag to link to newlib and the gcc support libraries.

These will be located at something like:

/usr/riscv64-elf/lib/rv32i/ilp32f/libc.a
/usr/riscv64-elf/lib/rv32i/ilp32f/libm.a
/usr/lib/gcc/riscv64-elf/12.1.0/rv32i/ilp32f/libgcc.a

Don't worry about the riscv64 as these are the 32 bit versions of the
libraries. Yours may indeed be at riscv32.

You'll also need a crt0.o file which provides the _start symbol,
initialises the stack pointer and jumps to main().

So something like:

riscv32-unknown-elf-gcc -fno-unroll-loops -O1 -fno-stack-protector -fno-pic
-march=rv32i -mabi=ilp32  \
/usr/riscv64-elf/lib/rv32i/ilp32f/libc.a \
/usr/riscv64-elf/lib/rv32i/ilp32f/libm.a \
/usr/lib/gcc/riscv64-elf/12.1.0/rv32i/ilp32f/libgcc.a \
-c -o test.o test.c

This doesn't provide crt0.o though, but it should get you further.

Alternatively, use theis compile script from Sylefeb as a basis?

https://github.com/sylefeb/Silice/blob/master/projects/ice-v/compile/icestick/compile_c.sh

ShervinShokouhi, 2022-05-16:
Thanks for your help, but where is "--specs=nano.specs" inside "https://github.com/sylefeb/Silice/blob/master/projects/ice-v/compile/icestick/compile_c.sh" script?
I used ice-v scripts but caused: undefined reference to __subsf3'

rob-ng15, 2022-05-16:
Did you include /usr/lib/gcc/riscv64-elf/12.1.0/rv32i/ilp32f/libgcc.a as
that should be in there?

rob-ng15, 2022-05-16:
Yes, I believe that the floating point emulation is in the libgcc.a file.

rob-ng15, 2022-05-16:
Just realised that I've given the paths to the versions that have hardware floating point support, I think you need to remove the f from the end of ip32

rob-ng15, 2022-05-16:
> Hi @ShervinShokouhi <https://github.com/ShervinShokouhi>, Hi @rob-ng15
>
> This missing entry is due to floating point emulation not being found by
> the linker. I never had to use it (I do fixed point / integer only) but we
> definitely need that to work. I'll try to look into the issue as soon as
> possible but @rob-ng15 <https://github.com/rob-ng15> has definitely more
> experience on this. Rob, I believe you did compile with floating point
> emulation in the past, right?

So the correct line is:

riscv32-unknown-elf-gcc -fno-unroll-loops -O1 -fno-stack-protector -fno-pic
-march=rv32i -mabi=ilp32  \
/usr/riscv64-elf/lib/rv32i/ilp32/libc.a \
/usr/riscv64-elf/lib/rv32i/ilp32/libm.a \
/usr/lib/gcc/riscv64-elf/12.1.0/rv32i/ilp32/libgcc.a \
-c -o test.o test.c

That was my error, the correct libraries are listed above. My CPU has
floating point support, so ilp32f is correct, without floating point it is
ilp32.

Try those libraries with the linker script and you should be fine.

Apologies.

ShervinShokouhi, 2022-05-16:
Hi @sylefeb, @rob-ng15 
Thanks for your detailed descriptions, now I'm able to compile C file to O. But in linker stage again asks for MATH functions:

C:\msys64\mingw64\bin\riscv64-unknown-elf-ld.exe: main.c:(.text+0x1e0): undefined reference to `__mulsf3'

How to link newlib library in linker stage?

rob-ng15, 2022-05-16:
Right, I didn't know that you were using Ming, I only use gcc. There must be an equivalent for Ming?

sylefeb, 2022-05-16:
I could not test beyond compilation yet but this seems to work on my side:
https://gist.github.com/sylefeb/64a0a55681f0680908989724c05e1778

Might need a change in paths, for libgcc in particular as it contains a version number (note: the order of object file / libraries does matter in the ld command line).

ShervinShokouhi, 2022-05-16:
I also compiled it :)
Thank you @sylefeb 
But I have some problem with test it in my custom SOC.
I'll write about results

ShervinShokouhi, 2022-05-16:
Another question: Now generated HEX file contains several sections split by something like "@000001E0", how do you fill them into BRAM for CPU?

sylefeb, 2022-05-16:
Hex file: I have updated the preprocessor script to do that: https://github.com/sylefeb/Silice/blob/draft/projects/ice-v/SOCs/pre_include_compiled.lua (could use a rewrite but does the job ;) ).

sylefeb, 2022-05-16:
> I'll write about results

Please let us know, hopefully we can make it work as well.

Btw, after a bit of searching, there is a simpler approach! The trick is to let gcc do the link because it knows where the libs are (found the hint at the bottom of the answer: https://stackoverflow.com/questions/64603070/risc-v-linker-cannot-find-lgcc ).

We can then use `-lm` (for math) and `-nostartfiles` (to use our custom crt0).

```
$ARCH-as -march=rv32i -mabi=ilp32 -o crt0.o $BASE/crt0.s
$ARCH-gcc -DICESTICK -nostartfiles -fno-builtin -fno-unroll-loops -O1 -fno-stack-protector -fno-pic -march=rv32i -mabi=ilp32 -T $BASE/config_c.ld -o $DST/code.elf $1 crt0.o -lm
```

That is way cleaner, I'll update my scripts accordingly.

ShervinShokouhi, 2022-05-17:
I compiled my code and tested it.
About your latest compilation script tested, I don't check it yet.
Newlib is fantastic, thank you @sylefeb.

sylefeb, 2022-05-18:
Thanks for the feedback and tests! I'll close the issue for now but feel free to reopen if needed.