sylefeb/Silice #221: 0.7 seconds delay after reset on the icestick.

at91rm9200, 2022-06-11:
Hello Sylvain,

I have tested the draft branch and find that it now takes about 0.7 seconds for a design to start up on the icestick. With the stable branch, it is in the microsecond range. Perhaps it has something to do with the changes in frameworks/boards/icestick.v?

```
algorithm main(output uint5 leds, inout uint8 pmod)
{
  uint26 cnt = 0;

  pmod.oenable := 8b11111111;
  
  while (1) {
    leds = cnt[21,5];   // first led output takes 0,7 seconds after reset.
    pmod.o = cnt[21,5];
    cnt  = cnt + 1;
  }
}
```
Regards, Bernd.

sylefeb, 2022-06-11:
Hello Bernd,

Indeed, this is due to this part of the framework (the RST counter): https://github.com/sylefeb/Silice/blob/7540cae853eceaedb62bc49f78bc0f888215f6ba/frameworks/boards/icestick/icestick.v#L156-L178

One issue on the ice40 is that BRAMs take time to initialize and as I encountered some problems I increased the delay very significantly, likely way too much! I'll refine this to make it as short as possible.

(If you need a quick workaround you can simply replace `icestick.v` by the one from the master branch, but I'll to look into this soon).

at91rm9200, 2022-06-11:
Hello Sylvain,

thank you for the explanation. I replaced icestick.v and everything is fine now.

Regards, Bernd.

closed 2022-06-11 by at91rm9200