sylefeb/Silice #222: How to set the subroutine permissions for the dot access of an algorithm input?

at91rm9200, 2022-06-19:
Hello Sylvain,

I am trying to set the subroutine permission to access "a1.value" in the algorithm "a" without success.

```
algorithm a(input uint1 value) <autorun>
{
   uint1 local_value = uninitialized;

   while (1) {
      local_value = value;
   }
}

algorithm main(output uint5 leds, inout uint8 pmod)
{
   a a1;

   subroutine s(writes _a1_value) {
                   //  ^ this should not be here

      a1.value = 1;
   }

   while (1) {
      () <- s <- ();
   }
} 
```
Without any permissions, the compiler realizes: "variable '_a1_value' is written by subroutine 's' without explicit permission". So I tried it with the permission "writes _a1_value", but this leads to the error: "^ this should not be here". Is this a bug?

Regards, Bernd.

sylefeb, 2022-06-19:
Hello Bernd,

There is a special `calls` keyword, which applies both to other subroutines and algorithm instances. I made an example [here](https://github.com/sylefeb/Silice/blob/draft/tests/alg_call_sub.si).

at91rm9200, 2022-06-19:
Hello Sylvain,

thank you. The permission "calls" indeed works.  I would not have guessed this, since I am using the dot access to the (auto running) algorithm and do not call it (I think).

Regards, Bernd.

sylefeb, 2022-06-20:
That is an excellent point, `calls` does not really match the usage here. I wonder, shall I add an extra (equivalent) keyword `uses`?

Regards,
Sylvain

at91rm9200, 2022-06-20:
Maybe the permission "uses" would fit a bit better than "calls" in this case.

But I am wondering, why the permission `writes` should not be the right one. Exactly this tells the Silice compiler, if I intentional omit the permission completely in the subroutine declaration: `variable '_a1_value' is written by subroutine 's' without explicit permission`. 

In my opinion, the subroutine should be declared like: 

   `subroutine s(writes a1.value)`

But I could be totally wrong.

sylefeb, 2022-06-21:
Indeed, it does make sense -- that might be either of `reads`/`writes`/`readwrites`, perhaps not having to specify each member, e.g. simply `writes a1`. Let's keep the issue open and I'll come back to it - thank you!

at91rm9200, 2023-04-08:
Hello Sylvain,

compiling the following code with the latest draft version produces a warning in line 7:
```
unit main(output uint5 leds, inout uint8 pmod)
{
   algorithm
   {
      subroutine sub(readwrites pmod)
      {
         pmod.o = 33; // <----- [warning] variable 'pmod_o' is written by subroutine 'sub' without explicit permission
      }

      while (1) {
         () <- sub <- ();
      }

   }
}
```
How can I avoid the warning?

With an earlier Silice version it was possible to declare the subroutine as follows:
```
subroutine sub(readwrites pmod_o)
```
But this leads now to an error: `cannot find referenced identifier 'pmod_o'...`

Regards, Bernd.

sylefeb, 2023-04-24:
Hi Bernd, thanks for the report, will look into it.

sylefeb, 2023-04-26:
This issue should be fixed in `draft` branch.

commit 134033ba (2024-01-25): fixes #222, regression on #103, cleanup in issue's tests
closed 2024-01-25 by sylefeb