sylefeb/Silice #223: Altera Cyclone IV EP4CE6E22C8N

ShervinShokouhi, 2022-06-22:
Hi to all

Any one here has experience with Altera Cyclone IV EP4CE6E22C8N?
How can I check my project resource consumption on for EP4CE6E22C8N?

Thanks

ShervinShokouhi, 2022-06-25:
Unable to recognise Quartus version via quartus_sh???

sylefeb, 2022-06-25:
Hi @ShervinShokouhi, 

There is no board with a CycloneIV in the repo ; there is however a well tested [framework for the de10nano](https://github.com/sylefeb/Silice/tree/master/frameworks/boards/de10nano) which is a CycloneV board. It should be possible to make a framework for your board, what kind is it? 

It's not just about the FPGA model but also the pinout and how it connects to peripherals, there is a small tutorial on adding a board here: https://github.com/sylefeb/Silice/blob/master/frameworks/boards/README.md

Normally, after creating the framework doing `make board` (with board the chosen name) should work. Edalize (which is used by Silice) will find the local Quartus installation and do the rest. I recommend starting by having the [blinky project](https://github.com/sylefeb/Silice/tree/master/projects/blinky) work.

Let me know how it goes -
Best regards,
Sylvain

sylefeb, 2022-06-25:
(regarding resource consumption, this will be available in the Quartus logs after synthesis)

ShervinShokouhi, 2022-07-01:
@sylefeb Thanks for your suggestions,
I'm trying to fit my board based on "de10nano" but some questions:
1) How setup "USB Blaster" as JTAG programmer. There is an error: "Error (213013): Programming hardware cable not detected"
    My settings are:
      "builders": [
      {
        "builder" : "edalize",
        "description": "Build using Edalize",
        "tool": "quartus",
        "tool_options": [
          {
            "family": "Cyclone IV",
            "device": "EP4CE6E22C8",
            "board_device_index" : 2
          }
        ],
        "toplevel" : "top",
        "bitstream"  : "build.sof",
        "constraints": [
          {"name": "build.sdc", "file_type": "SDC"},
          {"name": "pins.tcl", "file_type": "tclSource"}
        ],
        "program": [
          {"cmd" : "quartus_pgm", "args" : "-m jtag -o \"p;build.sof@2\""}
        ]
      }
      ]

2) If set up incorrect pin in pins tcl, for example setup incorrect pin for SDRAM:
    set_instance_assignment -name IO_STANDARD "3.3-V LVTTL" -to SDRAM_*

    Can this damage SDRAM chip?

sylefeb, 2022-07-01:
Hi - I believe programming is done either by Edalize or through Quartus (I don't precisely remember, will have to retest when I have the board in front of me). 

Regarding damage, I always assume bad things could happen and triple check everything. So far I never got into trouble, but I would expect that yes, there is the possibility of bad things happening if pins are improperly configured (most likely will be fine but the risk is not 0). The best would be to start from an example project and use the generated files as an example?

closed 2025-03-06 by sylefeb