sylefeb/Silice #225: Silice terminates without error message, if a subroutine contains two output parameters.

at91rm9200, 2022-06-25:
Hello Sylvain,

If the following design is compiled, Silice (draft branch) terminates without error message:
```
algorithm main(output uint5 leds, inout uint8 pmod)
{
   uint8 main_data1 = uninitialized;
   uint1 main_data2 = uninitialized;

   subroutine s(output uint8 data1, output uint1 data2)
   {
      data1 = 0;
      data2 = 1;
   }

   while (1) {
      (main_data1, main_data2) <- s <- ();
   }
}
```
It seems, as if the two output parameters of the subroutine `s` are the reason for this. 

Regards, Bernd.

sylefeb, 2022-07-01:
Hi Bernd,

Sorry for the delay in answering, thanks for the report! This looks rather serious, will fix asap.

Regards,
Sylvain

sylefeb, 2022-07-03:
Hi Bernd, this is fixed in all branches.

Thank you for the report -- good catch -- it was a depressingly simple copy-paste bug!

closed 2022-07-03 by sylefeb

at91rm9200, 2022-07-04:
Hello Sylvain,

thank you for fixing this. Runs great. :-)

Regards, Bernd.