sylefeb/Silice #232: Possible preprocessor problem.

at91rm9200, 2022-08-04:
Hello Sylvain,

the following source for the icestick results in a preprocessor error. This worked with an earlier Silice version from May 2022.

```
$$if ICESTICK then
   algorithm main(output uint5 leds, inout uint8 pmod)
$$else
   algorithm main(output uint5 leds)
$$end
{
  while (1) {
  }
}
```
```
silice-make.py -s blinky1.si -b icestick -p basic,pmod -o BUILD_icestick
* Source file                :  /home/bernd/fpga/silice/2208/Silice/mue/project/test_icestick/blinky1.si    [ok]
* Silice bin directory       :  /home/bernd/fpga/silice/2208/Silice/bin
* Build output directory     :  /home/bernd/fpga/silice/2208/Silice/mue/project/test_icestick/BUILD_icestick  (exists)
* Silice frameworks directory:  /home/bernd/fpga/silice/2208/Silice/frameworks [ok]
* boards description file    :  /home/bernd/fpga/silice/2208/Silice/frameworks/boards/boards.json 	[ok]
<<=- compiling blinky1.si for icestick -=>>
using default variant  configurable
using variant          configurable
using build system     edalize
assembling source /home/bernd/fpga/silice/2208/Silice/frameworks/libraries/memory_ports.si.
assembling source /home/bernd/fpga/silice/2208/Silice/mue/project/test_icestick/blinky1.si.
functionalizing unit main
error: [parser] Pre-processor directives are unbalanced within the unit, this is not supported.

```

Regards, Bernd.

sylefeb, 2022-08-05:
Hello Bernd,

Sorry, you stumbled on one of the very few breaking changes I had to make thus far (I try very hard not to introduce any). The fix is very simple but requires a code change:

```c
algorithm main(
$$if ICESTICK then
  output uint5 leds, inout uint8 pmod
$$else
  output uint5 leds
$$end
) {
  while (1) {
  }
}
```

The rule is that you can freely have preprocessor if-then-else between the '(' and ')' of the algorithm (or unit) in/out declaration, and between the '{' and '}' of the algorithm block. However there should not be an if-then-else enclosing and repeating these parentheses and braces.

I had to introduce this constraint when implementing the new preprocessor (that allows instantiation time preprocessing). There is a technical constraint there that the preprocessor has to be able to extract a 'simple' form of the algorithm or unit, free of top level if-then-else.

I was hopeful no one would encounter this breaking change, but you just prove me wrong ;-) I'll revise the documentation and try to clarify the error message.

Thanks for the report!
Best regards,
Sylvain

sylefeb, 2022-08-05:
New documentation section: https://github.com/sylefeb/Silice/blob/master/learn-silice/Documentation.md#preprocessor-constraints

at91rm9200, 2022-08-05:
Hello Sylvain,

thank you for the explanation.

Regards, Bernd.

closed 2022-08-05 by at91rm9200