sylefeb/Silice #240: Silice installs files back into the build directory

yurivict, 2023-01-08:
It installs these files:
```
/usr/ports/cad/silice/work/Silice-6a2beda/bin/silice
/usr/ports/cad/silice/work/Silice-6a2beda/lib/libantlr4-runtime.a
/usr/ports/cad/silice/work/Silice-6a2beda/lib/liblibsilice.a
/usr/ports/cad/silice/work/Silice-6a2beda/lib/liblua.a
/usr/ports/cad/silice/work/Silice-6a2beda/lib/libluabind.a
```

sylefeb, 2023-01-08:
Hi - thanks for bringing this up. I'll look into cleaning the libraries install in lib/ which should not be necessary. The install in bin was intentional to obtain a local install when checking out the repo, but I assume an install in the system dirs would be expected with possible customization of install paths from CMake?

yurivict, 2023-01-08:
It should be able to install into system folders.

Simple install(..) instructions should suffice for this.
The 'frameorks' folder should also be installed into /usr/local/share/silice/frameworks.

sylefeb, 2024-01-21:
Hi, I have finally revised the installation to follow standard paths (in `draft` branch). I will close this issue, but please refer to [this one](https://github.com/sylefeb/Silice/issues/255) for the next steps as it deals with the same problem. Thanks!

closed 2024-01-21 by sylefeb