sylefeb/Silice #243: Extracting bits from a bitfield results in a segmentation fault

rob-ng15, 2023-02-23:
Using a bitfield,

bitfield fp64{
    uint1   sign,
    uint11  exponent,
    uint52  fraction
}

and trying to select a part of one of the fields results in a segmentation fault.

e.g.

fp64( a ).exponent[0,8] causes a segmentation fault, whereas a[52,8] (effectively the same), works.

Will not build:
`uint64  aRET <:: df ? a : { 32hffffffff, fp64( a ).sign, fp64( a ).exponent[0,8], fp64( a ).fraction[29,23] };              // RETURN a ( as double ) or a ( as float repacked )
`

Works, effectively selecting the same bits:

`uint64  aRET <:: df ? a : { 32hffffffff, a[63,1], a[52,8], a[29,23] };                                                      // RETURN a ( as double ) or a ( as float repacked )
`

Not too much of an issue, but does help with code readability.

sylefeb, 2023-02-24:
Hi Rob, great catch, just pushed a fix. Thanks!

rob-ng15, 2023-02-27:
Cheers,

All seems to working fine. Nice quick fix, thank you!

closed 2023-02-27 by rob-ng15