sylefeb/Silice #244: Combining declaration and always assignment

rob-ng15, 2023-02-28:
Presently, a "variable" to be always assigned a value ( unless overwritten in algorithm ) has to be declared, then always assigned.

`
    uint1   OF = uninitialised;                                                                                                 // OVERFLOW FLAG
    uint1   UF = uninitialised;                                                                                                 // UNDERFLOW FLAG
    OF := 0; UF := 0;                                                                                                           // HOLD AT ZERO
`

Would it be possible/desirable to combine the declaration and always assignment?

e.g.

`unit OF := 0;`

My caution is that we already have =, <: and <:: as possibilities, but was thinking from a code readability point of view, easier to declare and always assign a value together when commenting.

Rob.

sylefeb, 2025-03-06:
Hi Rob, I believe this is addressed in #269? Please feel free to reopen if not. Thanks!

closed 2025-03-06 by sylefeb

rob-ng15, 2025-03-08:
Morning,

Just updated to the new draft branch and none of my projects will build...
PAWSv2 after a long time failed with...

Info:   at iteration #140: temp = 0.000074, timing cost = 74530, wirelen =
5303488
Info:   at iteration #145: temp = 0.000061, timing cost = 75146, wirelen =
5288118
Info:   at iteration #150: temp = 0.000047, timing cost = 73615, wirelen =
5274488
Info:   at iteration #155: temp = 0.000038, timing cost = 74645, wirelen =
5242743
Info:   at iteration #160: temp = 0.000030, timing cost = 74138, wirelen =
5211833
Info:   at iteration #165: temp = 0.000024, timing cost = 81325, wirelen =
5177017
Info:   at iteration #170: temp = 0.000020, timing cost = 80911, wirelen =
5136399
Info:   at iteration #175: temp = 0.000015, timing cost = 72846, wirelen =
5089399
Info:   at iteration #180: temp = 0.000012, timing cost = 75223, wirelen =
5042374
Info:   at iteration #185: temp = 0.000011, timing cost = 74934, wirelen =
4980661
Info:   at iteration #190: temp = 0.000009, timing cost = 66678, wirelen =
4905733
Info:   at iteration #195: temp = 0.000007, timing cost = 70883, wirelen =
4818650
Info:   at iteration #200: temp = 0.000006, timing cost = 68950, wirelen =
4736843
Info:   at iteration #205: temp = 0.000005, timing cost = 68040, wirelen =
4644882
Info:   at iteration #210: temp = 0.000004, timing cost = 62733, wirelen =
4545087
Info: Legalising relative constraints...
ERROR: failed to place chain starting at cell
'$nextpnr_CCU2C_1171$CCU2_COMB0'
ERROR: Placing design failed.
0 warnings, 2 errors

Rob.