sylefeb/Silice #246: Many boards share a common set of features, yet each board has its own define

sylefeb, 2023-03-05:
The example projects tend to be built for boards, using board-specific defines (ICEBREAKER, ULX3S) while many boards share a common FPGA and feature set (ice40, ecp5, etc.). Often only the pinout/clock differ. 

Should we get a more fine grained define system? This could be the opportunity to also get an improved (consistent) PLL system and revise/cleanup the pinout names...

rob-ng15, 2023-03-06:
Hello,

This does sound like a good idea, especially for adding extra boards, e.g.
ulx3s and ulx4m will be pretty similar, differing I/O pinouts, but the same
FPGA chip as you stated.

As for how, fpga.board.variant? So, the ulx3s would be ecp5.ulx3s.85f ?

Rob.

goran-mahovlic, 2023-03-07:
@sylefeb Thank you for accepting pull request for ulx4m! As for board versions on ulx4m for now we will have 

ulx4m-ls (ECP5 SDRAM) and ulx4m-ld (ECP5 DDR3) and each version will have revisions.

Goran