sylefeb/Silice #252: A subroutine which is never called in a design can crash Silice.

at91rm9200, 2023-05-27:
Hello,

while debugging a design, I commented out the call to a subroutine and happened to notice that Silice can crash in the process. The following shortened example shows the problem:
```
unit u(output uint1 data)
{
   algorithm {
      data = 0;
   }
}

unit main(output uint5 leds)
{

   uint1 result1 = uninitialized;
   u u1;

   algorithm
   {

      subroutine not_used (output uint1 sub_out, calls u1)
      {
         if (u1.data == 1) {
            sub_out = 1;
         }
      }

      while (1) {
         // (result1) <- not_used <- (); <--- subroutine is commented out.
         __display("%b",result1);
      }

   }
}

```
This is the console output:
```
bernd@Desktop:~/fpga/silice/2304/Silice/mue/project/test$ ./build.sh
silice-make.py -s test.si -b verilator -p basic -o BUILD_verilator
* Source file                :  /home/bernd/fpga/silice/2304/Silice/mue/project/test/test.si    [ok]
* Silice bin directory       :  /home/bernd/fpga/silice/2304/Silice/bin
* Build output directory     :  /home/bernd/fpga/silice/2304/Silice/mue/project/test/BUILD_verilator  (exists)
* Silice frameworks directory:  /home/bernd/fpga/silice/2304/Silice/frameworks [ok]
* boards description file    :  /home/bernd/fpga/silice/2304/Silice/frameworks/boards/boards.json 	[ok]
<<=- compiling test.si for verilator -=>>
using default variant  configurable
using variant          configurable
using build system     shell
launching command      /home/bernd/fpga/silice/2304/Silice/frameworks/boards/verilator/verilator.sh /home/bernd/fpga/silice/2304/Silice/mue/project/test/test.si
build script: SILICE_DIR     = /home/bernd/fpga/silice/2304/Silice/bin
build script: BUILD_DIR      = /home/bernd/fpga/silice/2304/Silice/mue/project/test/BUILD_verilator
build script: BOARD_DIR      = /home/bernd/fpga/silice/2304/Silice/frameworks/boards/verilator
build script: FRAMEWORKS_DIR = /home/bernd/fpga/silice/2304/Silice/frameworks
build script: FRAMEWORK_FILE = /home/bernd/fpga/silice/2304/Silice/frameworks/boards/verilator/verilator.v
VERILATOR_ROOT is set to 
using -Wno-TIMESCALEMOD
assembling source /home/bernd/fpga/silice/2304/Silice/frameworks/libraries/memory_ports.si.
assembling source /home/bernd/fpga/silice/2304/Silice/mue/project/test/test.si.
functionalizing unit u
functionalizing unit main
preprocessing 
                libraries_path = /home/bernd/fpga/silice/2304/Silice/frameworks/libraries
                framework_file = /home/bernd/fpga/silice/2304/Silice/frameworks/boards/verilator/verilator.v
              output_fsm_graph =                              1
simple_dualport_bram_wenable1_width =                              1
simple_dualport_bram_wenable1_type =                           uint
                bram_supported =                            yes
                 brom_template =              brom_generic.v.in
                 bram_template =              bram_generic.v.in
                brom_supported =                            yes
  dualport_bram_wenable1_width =                              1
             bram_wenable_type =                           uint
 simple_dualport_bram_template = simple_dualport_bram_generic.v.in
            bram_wenable_width =                              1
       dualport_bram_supported =                            yes
        dualport_bram_template =     dualport_bram_generic.v.in
                templates_path = /home/bernd/fpga/silice/2304/Silice/frameworks/templates
                frameworks_dir = /home/bernd/fpga/silice/2304/Silice/frameworks
  dualport_bram_wenable0_width =                              1
   dualport_bram_wenable0_type =                           uint
simple_dualport_bram_supported =                            yes
   dualport_bram_wenable1_type =                           uint
parsing algorithm main
instantiating unit 'u' as 'u1'
parsing algorithm u
algorithm u fsm 1b1397c0 num states: 2
 (no subs)
algorithm main fsm 1b05a650 num states: 3
**ASSERT FAILED** (if_deadend && !if_statless) || !if_deadend, file /home/bernd/build/Silice/006/Silice/src/Algorithm.cpp, line 4078

```
Regards, Bernd.

sylefeb, 2023-05-30:
Hi Bernd, fix in both `wip` and `draft`, thanks for the report!

Best regards,
Sylvain

at91rm9200, 2023-05-30:
Hello Sylvain,

thank you for the fix.

Regards, Bernd.

sylefeb, 2023-05-31:
Re-opening as my fix was actually incorrect. I pushed a correction in `wip` and `draft`, leaving open until further testing. Thanks to @rob-ng15 who detected the issue on PAWSv2!

at91rm9200, 2023-06-01:
Hello Sylvain, hello Rob,

I tested the draft version with two of my (larger) designs and both were working correct.

Regards, Bernd.

sylefeb, 2023-06-12:
Thanks for the tests! I found another corner case, fixed in my local branch, keeping open until next push.

commit 42257813 (2023-05-30): fixed #252, subroutines not called trigerring an assert on never reached states
closed 2024-01-25 by sylefeb