sylefeb/Silice #253: Not setting a subroutines output parameter leads to a yosys error.

at91rm9200, 2023-05-27:
Hello,

If a subroutine has an output parameter and this parameter is (accidentally) not set, then this leads to a Yosys error.

Here is an example:
```
unit main(output uint5 leds)
{
   uint1 result1 = uninitialized;

   algorithm
   {

      subroutine subr(output uint1 data_out)
      {
         // data_out = 1;
      }

      while (1) {
         (result1) <- subr <- ();
         __display("%b",result1);
      }

   }
}

```
This will eventually lead to the following Yosys error:
```
Generating RTLIL representation for module `\top'.

2.3.1. Analyzing design hierarchy..
Top module:  \top

2.3.2. Executing AST frontend in derive mode using pre-parsed AST for module `\M_main'.
Generating RTLIL representation for module `\M_main'.
build.v:322: ERROR: Identifier `\_q_o_subr_data_out' is implicitly declared and `default_nettype is set to none.
build.mk:9: recipe for target 'build.blif' failed
make[2]: *** [build.blif] Error 1
make[2]: Leaving directory '/home/bernd/fpga/silice/2304/Silice/mue/project/test/BUILD_icestick'

```
Maybe Silice could catch this situation?

Regards, Bernd.

sylefeb, 2023-05-30:
Hi Bernd, thank you for the report!

This was a bug and is now fixed (both `wip` and `draft` branch).

Best regards,
Sylvain

at91rm9200, 2023-05-30:
Hello Sylvain,

thank you for the fix.

Regards, Bernd.

closed 2023-05-30 by at91rm9200