sylefeb/Silice #255: Can't compile Silice on Arch Linux

Popolon, 2023-08-23:
I started to make an Arch Linux package of Silice, but with default compiler settings, compilation fail. 
* Version of GCC : gcc (GCC) 13.2.1 20230801

The error is the following one :
<pre>
/data/pacaur/silice/src/Silice/src/LuaPreProcessor.cpp: In member function 'void Silice::LuaPreProcessor::decomposeSource(const std::string&, std::map<int, std::pair<std::__cxx11::basic_string<char>, t_unit_loc> >&)':
/data/pacaur/silice/src/Silice/src/LuaPreProcessor.cpp:1058:51: error: format not a string literal and no format arguments [-Werror=format-security]
 1058 |           throw Fatal((w + " has no name").c_str()); // TODO: improve error report (line)
      |                                                   ^
/data/pacaur/silice/src/Silice/src/LuaPreProcessor.cpp: In function 'std::string fileAbsolutePath(std::string)':
/data/pacaur/silice/src/Silice/src/LuaPreProcessor.cpp:1263:11: warning: ignoring return value of 'char* realpath(const char*, char*)' declared with attribute 'warn_unused_result' [-Wunused-result]
 1263 |   realpath(f.c_str(), buf);
      |   ~~~~~~~~^~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors
make[2]: *** [CMakeFiles/libsilice.dir/build.make:116: CMakeFiles/libsilice.dir/src/LuaPreProcessor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:170: CMakeFiles/libsilice.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
</pre>

I didn't found the mean to resolve this problem in the code.

rob-ng15, 2023-08-23:
> /data/pacaur/silice/src/Silice/src/LuaPreProcessor.cpp: In member function 'void Silice::LuaPreProcessor::decomposeSource(const std::string&, std::map, t_unit_loc> >&)':

I'm on ARCH, I'll try tomorrow. Is this DRAFT branch.

Rob.

Popolon, 2023-08-23:
I used the master branch.

rob-ng15, 2023-08-24:
Morning,

I've just updated from github and rebuilt the "draft" branch. That is
building fine. I'll try the "master" branch in a moment.

Rob.

rob-ng15, 2023-08-24:
Just a quick update, "master" has built fine. It is possible that you are
missing a dependency for Silice to build. Check that you have lua, lua52,
luajit installed?

Rob.

sylefeb, 2023-08-24:
An ArchLinux package would be great! Thanks @rob-ng15 for testing ; on a fresh Archlinux (gcc 13.2.1 20230801) "master" also compiled on my side. @Popolon looking at the messages it seems like tighter security (warnings considered errors), maybe a different default g++ setting? I definitely will fix those, but would help to be able to reproduce on my side. I'll investigate!

sylefeb, 2023-08-24:
I pushed a fix (both in master and draft). I enabled the warning as error in the CMakeLists.txt (using -Werror=...) such that I can trigger the same compiler behavior. There might be others, please let me know how it goes.

(I'll likely enable -Werror to resolve all warnings, unfortunately a few are in third party libs).

Popolon, 2023-08-28:
@rob-ng15 I have both Lua version, but it's not related at all, that's an error related to type/cast, and strong default warning policy.

@sylefeb  Thank you very much, this resolved the problem. Everything is compiled. I will update this ticket with the informations about the Arch package, as soon I finished it.

Popolon, 2023-08-28:
The package is now available at : https://aur.archlinux.org/packages/silice-git

It looks like, from the sources, only silice binary is needed to be installed in bin/ can you confirm?

sylefeb, 2023-09-03:
Hi, sorry for the slow answer, indeed the Python scripts are needed to build, good catch for LibSL-small (this is because the verilator framework requires a compilation every time). The `frameworks` folder is also used by the build system, it contains board definitions, code templates and the verilator framework sources. If I understand correctly the PKGBUILD these might be missing currently?

However if you can successfully run verilator examples these are found somehow? To help see which folders are used, when a build occurs (eg. `make verilator` in `projects/vga_test`) the detected folders are displayed, see below:
[image]

Popolon, 2023-09-04:
I just tested to synthesize blink example in learn-silice folder and it worked with Verilator.

for framworks, I noticed in sources (confirmed  by a <code>strings silice| grep frameworks</code>  in binaries), that the path is ../frameworks, relative to the binary.

<pre>./src/SiliceCompiler.cpp:    frameworks_dir = std::string(LibSL::System::Application::executablePath()) + "../frameworks/";
./src/SiliceCompiler.cpp:  CONFIG.keyValues()["frameworks_dir"] = frameworks_dir;
./src/SiliceCompiler.cpp:  CONFIG.keyValues()["templates_path"] = frameworks_dir + "/templates";
./src/SiliceCompiler.cpp:  CONFIG.keyValues()["libraries_path"] = frameworks_dir + "/libraries";</pre>

In should be in <code>/usr/share/silice/frameworks</code> instead in a Linux Standard Base way. The pass can be given to the binary using <code>--frameworks_dir=</code> parameter, but it should probably be better to have this at compile time or a LSB way by default, on Linux platforms.

Popolon, 2023-09-04:
I just updated package to put frameworks in /usr/share/silice/frameworks and patch both silice binary and silice-make.py to have this as default path.

sylefeb, 2024-01-22:
Hi Everyone, this issue should be fixed by the changes in the `draft` branch. Everything is now installed in the usual `/usr/local/bin` and `/usr/local/share/silice` directories.

I also revised the 'getting started' scripts and created a `./get_started_linux.sh` script that I tested under Ubuntu (18, 20), Debian, Fedora, ArchLinux. 

I will merge to `main` very soon, please let me know if you see anything wrong.

sylefeb, 2025-03-06:
(closing as resolved, big merge with main branch upcoming)

closed 2025-03-06 by sylefeb

cheyao, 2025-07-03:
Leaving a comment here for anyone else who stumbles here in the future:

I've found that this problem existed because I was using clang as the compiler. Thus I just changed to gcc and the errors resolved themselves:
```
export CC=gcc
export CXX=g++
```