sylefeb/Silice #256: ULX3S sdram_memtest sample build fails.

gkankanh, 2023-11-08:
ERROR: cell type '$print' is unsupported (instantiated as '__main.sdram._q_sd_done_LUT4_B_Z_$print_EN')

Warnings: 1 unique messages, 1 total
End of script. Logfile hash: 24a6912103, CPU: user 1.00s system 0.05s
Yosys 0.35+7 (git sha1 5691cd095, aarch64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 18% 21x read_verilog (0 sec), 15% 35x opt_expr (0 sec), ...
nextpnr-ecp5 -l next.log --85k --freq 25 --package CABGA381 --timing-allow-fail --lpf /Users/gopal/FPGA/Silice/frameworks/boards/ulx3s/ulx3s.lpf --json build.json --textcfg build.config

gkankanh, 2023-11-08:
=== top ===

   Number of wires:                566
   Number of wire bits:           1785
   Number of public wires:         566
   Number of public wire bits:    1785
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                653
     $print                          1
     CCU2C                          56
     EHXPLLL                         1
     L6MUX21                        31
     LUT4                          282
     PFUMX                          55
     TRELLIS_FF                    211
     TRELLIS_IO                     16

gkankanh, 2023-11-08:
looks like it is the __display function that is non synthesizable

sylefeb, 2023-11-08:
Thanks - this is new behavior as __display was simply ignored in hardware as of Yosys 0.29+11. I've pushed a fix for this particular example (enclosing `__display` in pre-processor directives as it should have been). I'll look into a simpler approach where non synthesizable directives are simply ignored when targeting hardware.

gkankanh, 2023-11-08:
Thanks. Fantastic work. Really educative and inspiring. Keep it up.

closed 2023-11-08 by gkankanh