sylefeb/Silice #257: **ASSERT FAILED** (if_deadend && !if_statless) || !if_deadend, file ..Algorithm.cpp, line 4078

tommythorn, 2023-11-22:
I pulled out an old project, but it hits an assertion failure:
```
...
algorithm main fsm 128c0b0 num states: 2
algorithm main fsm 128c160 num states: 2
algorithm main fsm 128c210 num states: 2
**ASSERT FAILED** (if_deadend && !if_statless) || !if_deadend, file /Users/tommy/projects/Silice/src/Algorithm.cpp, line 4078
```

To reproduce:
```
$ git pull https://github.com/tommythorn/yarvi3/pull/new/assert-failure
$ make -C assert-failure
```
(the Makefile is set up for macOS)

sylefeb, 2023-11-25:
Thanks for the report! This was a missing check in implementing the revised pipelines. This now correctly reports an error message and a changelog warning.

The error is due to changes in pipeline syntax (see [CL0003](https://github.com/sylefeb/Silice/blob/master/ChangeLog.md#cl0003-pipeline-syntax) for details). To avoid this rewrite from

```c
while (1) {

  { A } -> { B } -> { C }

  if (...) { break; }

}
```

to 

```c
while (1) {

  { A -> B -> C }

  if (...) { break; }

}
```

That is because the `->` operator, which divides the design into stages, now has a lower priority than the blocks.

There have been other changes to pipelines, in particular those in `algorithm` blocks (the [documentation](https://github.com/sylefeb/Silice/blob/master/learn-silice/Documentation.md#pipelines) is up to date), so you may want to adjust your design a bit to be closer to the prior behavior. For your use case I'd recommend changing from

```c
algorithm main(output uint8 leds)
{
  always_after { A }
  while (1) {
    { S0 } ->  { S1 } ->  { S2 }
    if (...) { break; }
  }
}
```

to

```c
unit main(output uint8 leds)
{
  always {
    { S0 -> S1 -> S2 }
    if (...) { __finish(); /*simulation only*/ }
    A
  }
}
```

This should be equivalent to the prior version.

Hope that helps!
Best regards -

sylefeb, 2023-12-30:
(Closing as solved for now, please let me know if the problem persists)

closed 2023-12-30 by sylefeb

tommythorn, 2024-01-03:
Sorry.  Thank you for the fix which solved the crash I experienced.  However the resulting design didn't work as expected and I timed out understanding why and failed to update the github issue.