sylefeb/Silice #270: Unable to build just the ice-v CPU

MrJake222, 2024-05-31:
Hello, I'd like to use ice-v CPU core and integrate it into existing Verilog infrastructure. I'm new to Silice. Trying my best but I keep getting:
```
$ silice \
--export rv32i_cpu \
--frameworks_dir /home/xxx/git/Silice/frameworks \
-f '/home/xxx/git/Silice/frameworks/boards/icebreaker/icebreaker.v' \
CPUs/ice-v.si 

assembling source /home/xxx/git/Silice/frameworks/libraries/memory_ports.si.
assembling source CPUs/ice-v.si.
functionalizing unit execute
functionalizing unit rv32i_cpu
preprocessing 
                libraries_path = /home/xxx/git/Silice/frameworks/libraries
                framework_file = /home/xxx/git/Silice/frameworks/boards/icebreaker/icebreaker.v
              output_fsm_graph =                              1
            __finish_supported =                             no
           __display_supported =                             no
simple_dualport_bram_wenable1_width =                              1
simple_dualport_bram_wenable1_type =                           uint
                bram_supported =                            yes
                 brom_template =              brom_generic.v.in
                 bram_template =              bram_generic.v.in
                brom_supported =                            yes
  dualport_bram_wenable1_width =                              1
             __write_supported =                             no
             bram_wenable_type =                           uint
 simple_dualport_bram_template = simple_dualport_bram_generic.v.in
            bram_wenable_width =                              1
       dualport_bram_supported =                            yes
        dualport_bram_template =     dualport_bram_generic.v.in
                templates_path = /home/xxx/git/Silice/frameworks/templates
                frameworks_dir = /home/xxx/git/Silice/frameworks
  dualport_bram_wenable0_width =                              1
simple_dualport_bram_wenable0_width =                              1
   dualport_bram_wenable0_type =                           uint
simple_dualport_bram_supported =                            yes
   dualport_bram_wenable1_type =                           uint
[preprocessor] 165]  attempt to concatenate a nil value (global 'addrW')
error: the preprocessor was interrupted
```

Any suggestions? I may be asking in the wrong place, I know. If so, please point me in the right directions (you guys have a forum?)

sylefeb, 2024-06-01:
Hi, this is the right place to ask! Exporting is not well document but absolutely possible (an example is in `projects/qpsram/export/`. I'll help you doing this for the ice-v. I assume you need only the CPU, not the SOC around? Which variant of the processor where you considering?

sylefeb, 2024-06-01:
Just pushed an example in `projects/ice-v/export`, running `make` will produce the file `M_icev_cpu.v` which contains the CPU as module `M_icev_cpu`. The memory interface is for a BRAM-like memory that answers read/writes in one cycle.

sylefeb, 2024-07-10:
Hi @MrJake222,

May I close this issue? is it resolved?

Thanks!

sylefeb, 2025-03-05:
(closing as resolved)

closed 2025-03-05 by sylefeb