sylefeb/Silice #271: silice-make.py fails on initial test after install on MacOS

resistor, 2024-06-03:
I installed silice per the instructions here: https://github.com/sylefeb/Silice/blob/master/GetStarted_macOS.md

When I to build the `divstd_bare` project using the instructions from [here](https://github.com/sylefeb/Silice/blob/master/GetStarted.md#testing), I encounter multiple issues.

First, I get a permissions error on `silice-make.py`. I manually fixed them since it was obvious what needed to be changed.

However, now the build just failed with the following output:

```
divstd_bare % make icarus   
silice-make.py -s main.si -b icarus -p basic -o BUILD_icarus  
* Source file                :  /Users/resistor/Documents/Silice/projects/divstd_bare/main.si    [ok]
* Silice bin directory       :  /Users/resistor/install-silice/bin
* Build output directory     :  /Users/resistor/Documents/Silice/projects/divstd_bare/BUILD_icarus  (exists)
* Silice frameworks directory: 			[not found]
make: *** [icarus] Error 255
```

sylefeb, 2024-06-03:
Hi - sorry you're having difficulties with this. I cannot easily test on MacOS so things are less polished there.  What seems to be happening is that the Silice executable is not able to locate files it needs to compile the design.

Silice now needs to be installed on the target system, running `make install`. Once this is done, the framework files should be in the equivalent of `/usr/local/share/`, subdirectory `silice`. Could you check that this is the case? It seems your install is in `/Users/resistor/install-silice/`, did you specify this using `-DCMAKE_INSTALL_PREFIX=<path>` on cmake? See e.g. 

https://github.com/sylefeb/Silice/blob/364b63b4eba9924fa07f360dc69e28ceeb9765f9/compile_silice_mingw64.sh#L28

This path gets passed to the source code by cmake to locate the files: https://github.com/sylefeb/Silice/blob/364b63b4eba9924fa07f360dc69e28ceeb9765f9/CMakeLists.txt#L164

resistor, 2024-06-03:
Yes, I set a custom prefix using `CMAKE_INSTALL_PREFIX`.

I see the following in `flags.make`: 
```
CXX_DEFINES = -DANTLR4CPP_STATIC -DFRAMEWORKS_DEFAULT_PATH=\"/Users/resistor/install-silice/share/silice/frameworks\"
```

It looks like there's an incorrect trailing `\`?

resistor, 2024-06-03:
Here is my exact CMake invocation:
```
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/Users/resistor/install-silice -G "Unix Makefiles" ../..
make -j16
make install
```

The `share` directory does appear to be installed correctly:
```
% ls /Users/resistor/install-silice/share/silice/frameworks/
boards		libraries	templates	verilator
```

closed 2024-06-23 by sylefeb